Ultimaker/Cura #10442: hi.whenever i want to import stl files,cura crashes and close.version 4.1.1.0-amd64

Gieterkie64, 2021-09-13:
### Application Version

4.9.0 + 4.1.1.0

### Platform

Windows 10

### Printer

creality CR10

### Reproduction steps

nothing

### Actual results

NA

### Expected results

Files should have been imported

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

NA

nallath, 2021-09-13:
If you report a bug, please include the log files.

DO you have chitubox installed? If so, it's probably this issue: https://support.ultimaker.com/hc/en-us/articles/4402996140946-Ultimaker-Cura-crashes-when-opening-any-file

Gieterkie64, 2021-09-13:
Yes I have Chitubox installed for my resin printer. I overcame the problem by going to my STL folder and click on "open with" and then click on Cura

Gieterkie64, 2021-09-13:
Its ok .I'll just use Prusa or Craftware slicer. They work with Chitubox installed on my PC

nallath, 2021-09-13:
If you read the instructions, you can see how to get both to run side by side.

If you want this fixed properly, please raise the issue with Chitubox, since they are the ones that broke this for other applications (including Cura)

closed 2021-09-13 by nallath