Ultimaker/Cura #10444: Cura 4.11 makes odd nonprinting move

bill-orange, 2021-09-13:
### Application Version

4.11

### Platform

Windows 10

### Printer

Anet AM8

### Reproduction steps

1. slice two small parts
2.  observe bed boundary violation (see line 6 of gcode)

### Actual results

1. Slice parts
2. Odd out of bounds non-printing movement resulting in boundary violations (see line 6 of gcode).

### Expected results

No boundary violation

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[Cura_miscalculation.zip](https://github.com/Ultimaker/Cura/files/7155352/Cura_miscalculation.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/7155365/cura.log)

GregValiant, 2021-09-13:
Well @bill-orange I do my little detective work here and through most of it I've found that the problems are often elsewhere and not the fault of Cura.  And then one like yours comes up.  It's interesting.  Your Gcode file has these lines in it:
_;MINY:-1.14668e+06_

G1 F3900 E213.41833
;MESH:NONMESH
G0 F600 X117.161 Y93.403 Z3.88
G0 F12000 **X373.89 Y-1146676.546**  The location is not on the build plate
;TIME_ELAPSED:6488.478852
;LAYER:31
;MESH:restraint_RG_316.stl
G0 X373.89 Y-1146676.546
;TYPE:WALL-INNER
G1 F3900 **E216.41833**
G1 F2400 X104.814 Y101.55 **E18839.24758**  The extrusion amount is exceedingly large.  

Here's what I found.
First I deleted the Small Radius part and the resulting slice was fine (clue 1).  Then I deleted the large radius part and the error showed up.  Rotating the small radius part eliminated the problem (clue 2).  The clues pointed to a problem with the model.

I exported the parts from Cura as separate STL's.  I opened each file in MS 3D Builder.  The large radius part was fine, but the small radius part immediately threw up a red flag.  The program was able to fix the file.  Cura's Mesh Tools did not find any problem with the part.  So I tried a third way to break the tie.
This is the report from https://formware.co/OnlineStlRepair that I also used for the repair.  You can see that not much was found to be wrong, but the vertexes and triangles changed.
--> 0 Naked edges (?)
--> 0 Planar holes (?)
--> 0 Non-planar holes (?)
--> 0 Non-manifold edges (?)
--> 0 Inverted faces (?)
--> 0 Degenerate faces (?)
--> 0 Duplicate faces (?)
--> 0 Disjoint shells (?)
-> Repairing: 100.00%
----- Repair completed in 2876ms ------
-> Vertex count changed from 6726 to 6771 (+45)
-> Triangle count changed from 13452 to 13542 (+90)
Yes, there is a slight problem with one of the parts and it was causing the problem with the out-of-whack Y values.
Slicing the model with the error resulted in huge moves to the positive Y as well as negative.  The positive moves were not noticed by Cura other than in the preview.  That might be a display issue.

On another note, but still in regards to illegal movement:
In your ending Gcode you have a G92 putting the printer into Relative Mode and then you have a line "G0 X0 Y205 ;move X/Y to min endstops,".  That becomes a relative move and the printer attempts to go (in this print) to Y300 because it is starting out at Y95.  I suggest you rearrange the ending of your End Gcode to:
G91 ;relative positioning
G1 E-1 F300  ;retract the filament a bit before lifting the nozzle, to release some of the pressure
G1 Z+0.5 E-5 X-20 Y-20 F9000 ;move Z up a bit and retract filament even more
**G90 ;absolute positioning**﻿
G0 X0 **Y{machine_depth}** ;move X to min and Y to max, so the head is out of the way
M84 **X Y E** ;all steppers off except Z (because some heads drift down)

The attached Zip file contains both parts as STL's and the small radius part is the repaired version.
[Cura_Restraint_GV.zip](https://github.com/Ultimaker/Cura/files/7157342/Cura_Restraint_GV.zip)

bill-orange, 2021-09-13:
Thanks for the feedback. I was aware of the end code problem and I thought that I fixed it.  Since I often use older project files for their profile, old problems keep coming up.

Interesting that such a small error in the model (exported from Fusion360) caused the calculation error.  I ended up slightly revising both parts. They are intended to grip coax cable and the grip was too weak.  The revised parts did not have the problem, adding to the mystery.  

I will close this issue.  If the same sort of thing happens to other users in a wider range of situations, it can always be reopened.

closed 2021-09-13 by bill-orange

Ghostkeeper, 2021-09-21:
I wouldn't expect that a broken model like that would cause such huge g-code coordinates though. Maybe that's a bug in clipper? We've seen before that Clipper can create some huge coordinates in some edge cases.

When I try to slice the (original) project file, I'm not getting any big travel moves. The only move outside of the build volume was the one from the end g-code.

bill-orange, 2021-09-21:
Did you mean Fusion360 rather than Klipper?  Since the problem appeared in the Gcode, Klipper did not have a chance to screw it up yet.

Ghostkeeper, 2021-09-28:
No, I meant [Clipper](http://angusj.com/delphi/clipper.php), not Klipper.