Ultimaker/Cura #10447: 4.11.0 - Dual extruder behavior is off

tlhintoq, 2021-09-14:
### Application Version

4.11.0 release

### Platform

Win10x64

### Printer

Hictop D3 Hero (IDEX)

### Reproduction steps

1. Bring in two objects.
2. Place object 1 on the bed (z=0)
3. Place object 2 on top of object 1 (z = 1.5 or whatever)
4. Select Extruder 2 for Object 1 (bottom object to print from Right extruder)
5. Select Extruder 1 for Object 2 (raised object to print from Left extruder)
6. All extruder choices remain at "not overridden" except for bed adhesion because you have to select an extruder
7. Slice

### Actual results

The sliced file keeps trying to pre-heat the unused extruder.
The second object that is starting at Z=1.5 still tries to print a brim on the bed even though that object's extruder is not selected as the bed adhesion extruder.

Look at the start of this layer:7, which is still only object 1 from extruder2/T1.  Extruder1/T0 has nothing to do here so why has command switched to T0?
It specifies a switch to T0, then heats T1. For the object on T1, with a brim on T1. But control was switched to T0.  Huh?

;LAYER:7
G1 F1200 E175.16806
G1 F600 Z2.64
G92 E0
T0
G92 E0
M105
M109 S215
M104 T1 S230
G1 F1200 E12
G1 F600 Z1.64

### Expected results

Second object at Z>0 shouldn't be trying to print bed adhesion at all.
Don't pre-heat an extruder that isn't going to be used.
Don't switch control to the wrong extruder.

Cura .3mf and .gcode attached in the zip
[image]
[1xshell_brassRight.zip](https://github.com/Ultimaker/Cura/files/7158119/1xshell_brassRight.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

No additional info: Covered in the fields above.

tlhintoq, 2021-09-14:
Oh... And everyplace its erroneously changing from T1 to T0... Its also sending a G92 E0 to reset the extruder counter.
Not going to make hand scrubbing through the gCode very easy.

tlhintoq, 2021-09-14:
[image]

Erabior, 2021-09-15:
I have a similar problem where on the Vivedino Trex 3.0 If I want to use Extruder 2 and only extruder 2, in several cases, the printer will properly heat E2 but attempt to print with E1

Ghostkeeper, 2021-09-21:
> [1xshell_brassRight.zip](https://github.com/Ultimaker/Cura/files/7158119/1xshell_brassRight.zip)

This project file uses a printer definition that doesn't exist, "d3". I think you modded that into Cura? Could you share the files you added for this printer?

I think the reason it's adding that heat command is because it will prime all extruders on the first layer using the extra skin lines around the brim. Aside from adding adhesion, one of the functions of the brim is to prime. This needs to happen with all extruders that are used anywhere in the print.
You can prevent that by setting the Skirt/Brim Minimum Length to 0 for the extruder you don't want those lines with.

closed 2021-10-05 by no-response[bot]