Ultimaker/Cura #10450: How about a nice duplicate tool?

tlhintoq, 2021-09-14:
### Is your feature request related to a problem?

If you duplicate an object in any other slicer, you get a nice grid of objects.  Make 15 more of something and you tend to get a 3x5 grid per the spacing preferences you have set.

Not in Cura.  In cura you get a mess.
[image]

### Describe the solution you'd like

It would be nice to have some spacing options... Like... gap X, gap Y... Then when you do a cntrl-m duplicate of an object and make 20 of them... That they naturally fall into a decent grid following the aspect ratio of the bed.

### Describe alternatives you've considered

Maybe the popup could have spacing or rowXcolumn duplication.
[ 3 ] Rows by
[ 5 ] Columns
[ x ] Start in center

### Affected users and/or printers

All

nallath, 2021-09-15:
Duplicate of #7284

nallath, 2021-09-15:
The other part of the request is mentioned in https://github.com/Ultimaker/Cura/issues/8621

closed 2021-09-15 by nallath