Ultimaker/Cura #10451: (4.11.0) Slicing Failed

meighen8907, 2021-09-14:
### Application Version

4.11.0

### Platform

WIndows 10

### Printer

Ender 3 V2

### Reproduction steps

1. Chose the "3D Print" button in Fusion 360 after making some edits to the model.
2. Cura opened up like normal with the STL loaded.
3. Adjusted the print settings to use supports.
4. Clicked on Slice

### Actual results

1. It didn't slice. 
2. Produced a "slicing failed" message
 
[image]

### Expected results

It should have sliced.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[Body1.zip](https://github.com/Ultimaker/Cura/files/7163623/Body1.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/7163683/cura.log)

meighen8907, 2021-09-14:
I was finally able to get it to slice by doing the following steps.
1. Once STL is loaded in Cura, I selected the top face to align to the build plate so it basically flipped it upside down.
2. Then I did the same thing but for the bottom. Selected the bottom face (now on the top) to align to the build plate.
3. It now slices.

I have to do this each time I load this STL into Cura to print it.

nallath, 2021-09-15:
If you just "jiggle" the object a bit, it also works (eg; just move it a tiny bit in the x or y direction). 

Based on the logs I can see that it's crashing on a bad memory allocation. I will discuss this with the rest of the team and keep you up to date!

Ghostkeeper, 2021-09-20:
This is a duplicate of https://github.com/Ultimaker/Cura/issues/10418.

Disabling the new monotonic skin or changing the top/bottom pattern from Zigzag to Lines works as a workaround. Or, indeed, moving the model slightly such that this edge case doesn't occur any more. Skin lines are not (properly) moving along with the model, so moving the model moves the relative position of skin lines such that the cases of zigzag joinage are different.

closed 2021-09-20 by Ghostkeeper