Ultimaker/Cura #10454: slicing problem

Gramkracker, 2021-09-14:
### Application Version

4.11.0

### Platform

windows 10

### Printer

ender 3

### Reproduction steps

imported a file tried to slice and it showed a error

### Actual results

i tried multiple times to slice keeps saying error

### Expected results

to slice file

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CE3_Body_0.zip](https://github.com/Ultimaker/Cura/files/7164415/CE3_Body_0.zip)

nallath, 2021-09-15:
First off, it seems that your model itself has some issues with it: [image]

I found that when slicing this with default settings, it did work. So I narrowed it down to "top/bottom pattern" being set to Zigzag as the root of this problem (I don't know why it's doing that though).

I will discuss this with the rest of the team!

GregValiant, 2021-09-16:
The on-line repair service I use was unable to repair this model.  MS 3D Builder and IdeaMaker made some progress, but I couldn't get it to slice in Cura and in fact it crashed Windows hard enough to require a power off and restart.  Twice.

Ghostkeeper, 2021-09-21:
This is a duplicate of https://github.com/Ultimaker/Cura/issues/10418.

Jiggling the model slightly sometimes fixes it. But the easiest workaround is to use the Lines pattern rather than Zigzag, for the top/bottom.

closed 2021-09-21 by Ghostkeeper