Ultimaker/Cura #10456: MONITOR PRINT JOB MENU IS CUTOFF:   Cannot cancel jobs in Cura 4.11 with Ultimaker 3.

polygonfuture, 2021-09-15:
### Application Version

4.11

### Platform

Windows 10 

### Printer

Ultimaker 3

### Reproduction steps

1.  Started a print job.
2. Go to monitor page.
3. Print jobs are cut off and I am unable to edit or change jobs.

Going to the web portal based print queue there are no settings to edit or cancel jobs either.     

[image]

### Actual results

I cannot cancel or edit a job because the options are cutoff.  

[image]

### Expected results

The entire menu should be visible.  This is very QC.  

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[problemProject.zip](https://github.com/Ultimaker/Cura/files/7166334/problemProject.zip)

nallath, 2021-09-15:
This is one of the things that is tested in our regression, so no worries. Our QC team checks this ;)

I can also not reproduce this on my system:
[image]

Could you provide a bit more information about your system? Do you have a different text size? What resolution do you use?

polygonfuture, 2021-09-15:
Machine is a 16" 16:10 screen laptop running 2560x1600 resolution at 125% scaling.    Not other changes have happened to the machines text beyond windows scaling.

Ghostkeeper, 2021-09-22:
Indeed it doesn't reproduce for me either, not even by setting the screen scaling to 125% or 200%. I also tried setting just the font size bigger in my operating system, not the actual screen scale.
[image]

I briefly looked through the `MonitorPrintJobCard.qml` file that seems to be the issue here, and I think the problem could still be that some parts of the interface are defined as pixels multiplied by the screen scale, and some parts are defined from the theme (which uses font size - EMs - to measure things). I wasn't able to reproduce the issue but I only tried the accessibility option to increase font size in Ubuntu which doubles font size. I didn't really do an exhaustive test there.