Ultimaker/Cura #10459: Skin Overlap does not apply to Top Surface Skin Layers

geoffrey-young, 2021-09-15:
### Application Version

4.11.0

### Platform

OSX

### Printer

Voron 2

### Reproduction steps

slice a model with some perimeters, with at least one Top Surface Skin Layer, and an unusually large Skin Overlap value

### Actual results

the actual overlap of the Top Surface Skin Layer with the walls remains unchanged no matter what the value of Skin Overlap.  even if walls are not considered skin, this happens with or without Extra Skin Wall Count > 1.  for reference, all Top Layers that are not Top Surface Skin Layers behave appropriately.

### Expected results

the Top Surface Skin Layers should overlap in proportion to the Skin Overlap value, equal to the overlap that occurs in the non-Top Surface Skin Layers

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[2mm_washer.3mf.zip](https://github.com/Ultimaker/Cura/fil
[image]
[image]

konskarm, 2021-09-21:
Hi @geoffrey-young. Thank you for the report. I can reproduce the issue so I'll bring it up with the team for discussion.

evtrados, 2021-09-22:
Hello @geoffrey-young,

We discussed this issue with the team and we already had it on our backlog for some time. For the Cura devs, please check the internal ticket CURA-6550.

Ghostkeeper, 2021-09-22:
That makes this a duplicate of https://github.com/Ultimaker/Cura/issues/5758 then.

closed 2021-09-22 by Ghostkeeper