Ultimaker/Cura #10462: Prime Blob problem on UM S5 Pro

TheDesignerDan, 2021-09-16:
### Application Version

4.9.0

### Platform

Windows 10

### Printer

Ultimaker S5 Pro Bundle

### Reproduction steps

1. Set up a print using PLA in nozzle 1 and Breakaway in Nozzle 2.
2. "Enable Prime Blob" in Build Plate Adhesion settings.

### Actual results

The problem seems worse for the Breakaway in Nozzle 2 than for the PLA in nozzle 1, but they both do this.

1. The nozzle creates the prime blob and lifts up to accommodate the material beneath it.
[image]
2. As the nozzle moves to the centre of the buildplate to start printing, it immediately starts to lower to the level of the buildplate.
[image]
3. This squashes the prime blob under it, smearing it along with the printer head thereby sticking to the head and often to the resulting print.
[image]
[image]

### Expected results

Ideally, the print head should stay at the higher level until it has moved far enough away from the prime blob so that it won't smear it.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

It does this with every print.

nallath, 2021-09-16:
The actual behavior of what the printer does when it gets the G280 (prime) isn't something that Cura controls. So if you want to notify that team, I'm going to have to point you to the generic support link again (Sorry!):  https://support.ultimaker.com/hc/en-us/requests/new

But by default the prime blob for the S5 is disabled. In order to ensure that the priming happens, I can recommend enabling a skirt or a brim.

TheDesignerDan, 2021-09-17:
Good morning Jamie
Thanks for your reply.

Kind Regards,

Daniel Lehmann
Product Designer: Design First
Polyoak Packaging (Pty) Ltd

closed 2021-09-23 by Ghostkeeper