Ultimaker/Cura #10464: Lay brim from inside out......

PhilBaz, 2021-09-16:
### Is your feature request related to a problem?

Yes, when bad adhesion leads to high squish for first layer (higher than would be ideal otherwise),  Subsequent brim rings can produce a wavy edge. Often the wavy edge develops on the third of fourth ring.

This can impact the surface skin.

### Describe the solution you'd like

If Brims were laid from the inside out the wavy edge would be on the outside and would not impact surface skin.

### Describe alternatives you've considered

 If you need a brim, and bad adhesion leads you to squish the first layer until its wavy, then there is currently no solution.

### Affected users and/or printers

all

### Additional information & file uploads

[image]

nallath, 2021-09-17:
I think this is a pretty good suggestion because there is another advantage that you didn't list; When printing the inside first, you also give the parts that touch the model more time to cool down, which should decrease the warping.

rburema, 2021-09-17:
(As an aside, you could also set the 'Brim Gap' higher than 0 (to a fraction of the line-width). That doesn't solve the full problem, but might be a workaround for a lot of cases.)

nallath, 2021-09-17:
I've just discussed this with one of our process & material engineers. He did come up with a good point that might be an issue for this. The first line can be a bit under extruded (or not extruded at all). If that is the case, the brim will not be stuck to the object at all if it's printed inside to outside.

PhilBaz, 2021-09-17:
Could we use a negative brim offset/gap to address this?

PhilBaz, 2021-09-19:
Just had another thought on this. Perhaps adding a note in the flyout info
that it's best used with a prime line in the startup G-code. I think a
prime line or something similar is the widely used best practice at this
point, but then it covers the issue in a sense. I do think it would be a
valuable addition.

Ghostkeeper, 2021-09-24:
First rule of UX design is that almost nobody reads text :wink: