Ultimaker/Cura #10468: (4.11.0) Cannot add cloud printer

solarstandard, 2021-09-17:
### Application Version

4.11.0

### Platform

Windows 10

### Printer

Ultimaker 3E

### Reproduction steps

1. Click the printer menu
2. Click Add printer
3. Click Add cloud printer

### Actual results

The Add a Cloud printer ialog is stuck in the Waiting for Cloud response phase indefinitely. The printer is connected to the Digital Factory, I can ping it from the Maintenance menu there.

### Expected results

The cloud printer is added.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Cura log attached: [cura.zip](https://github.com/Ultimaker/Cura/files/7183035/cura.zip)

lf-axentia, 2021-09-17:
I experience the same thing.
One of my cloud printers (S3) are missing in Cura 4.11.0. And "Add a Cloud printer" waits forever. My other printer (S5) is automatically available in Cura after signing in to my Ultimaker account.

Both of my printers (S5 and S3) are available in Digital factory, Cura 4.10.0 and Cura Arachne beta from the same computer. For me this problem is isolated to 4.11.0.

evtrados, 2021-09-22:
Hi @solarstandard and @lf-axentia ,

I could indeed reproduce the issue you guys are facing and could backtrace it to Cura v4.7.1. It seems that when using the "Add Cloud printer" button there's no request being made to the cloud at that exact moment to fetch any remaining available printers.
A workaround I found was to go to your account icon on the top right and click "Check for account updates", as this will generate a request and your printer will get added (or at least mine did).
Furthermore, I know that we make a sync request every 60 seconds or so, which means that if you wait for that amount of time after pressing "Add cloud printer", it will eventually be found and displayed.

In any case, I'll bring this up with the rest of the team and will create a bug ticket out of it.

lf-axentia, 2021-09-22:
Hi @evtrados
I'm happy to hear that you managed to backtrack.
Unfortunately the workaround "Check for updates" is not working for me. I get a new timestamp of last update. But my lost cloud printer is not showning up.
To me this "All cloud printer is not visible"-problem was introduced in Cura 4.11.

evtrados, 2021-09-23:
Hmmm, that indicates a potential issue elsewhere then. Pardon my silly question, but could it be that this printer is currently offline? We can only reproduce such behavior for offline printers, as even though they are added to the digital factory, the fact that they're offline makes them inactive and, thus, non-discoverable by Cura. Could you please verify for me that you can see the printer if you go to https://digitalfactory.ultimaker.com/app/printers ?

If the printer is online and visible in the digital factory we will have to dig deeper to find the root cause of this issue.

lf-axentia, 2021-09-23:
Silly question first. Yes, both printers are online. Available at  [ https://digitalfactory.ultimaker.com/app/printers](https://digitalfactory.ultimaker.com/app/printers) and starts flashing on sending them "Identify command"
[image]
Same account on Cura 4.11.0 Problem. Only one printer available.
[image]
Same account on Cura 4.10.0 Expected result. Both printers available
[image]

Please dig deeper. Is there any relevant logs I may provide?

evtrados, 2021-09-23:
Thanks for the detailed report! That looks very weird, indeed.
We're going to discuss this with the team and see what we can do about it.

In the meantime it would be nice if you could send us the following, so that we check if we can reproduce it on our end:

- your whole configuration folder. That would be the folder called `cura` within the `C:\Users\%USERNAME%\AppData\Roaming` directory
- a logfile after you try and sign in on your Cura 4.11. That logfile is the `stderr.log` file which resides within the `cura` folder mentioned above.

lf-axentia, 2021-09-23:
Here in the requested files.
[stderr.log](https://github.com/Ultimaker/Cura/files/7217337/stderr.log)
[curaconfig.zip](https://github.com/Ultimaker/Cura/files/7217426/curaconfig.zip)

konskarm, 2021-09-23:
Hi @lf-axentia. This is quite a strange issue indeed as we cannot reproduce it locally. The logs don't provide any information that can help us identify the issue either. 
Could you try deleting all the contents of the 4.11 config folder and try logging in again (just make sure to keep a copy of the folder before deleting everything in case you want to restore it)?

**Note:** Deleting all the contents of the 4.11 without deleting the folder will work as a fresh install of Cura. If you also delete the 4.11 folder, it will inherit things from 4.10 which may prevent us from identifying something.

lf-axentia, 2021-09-23:
All printers are available in Cura 4.11.0 after restarting with:
- inherited configuration. No folder named "4.11".
- no configuration. Folder "4.11" left empty.

With my provided, broken configuration folder the issue persists. Strange that you could not reproduce. At first my configuration was inherited from 4.10 but I don't remember if the printer was lost in the update och afterwards.

I stay with the fresh configuration, I think.

konskarm, 2021-09-23:
@lf-axentia I'm glad that the fresh install helps. This means that it is an isolated insident and not something inherently wrong in Cura.

I tried with your configuration folder but there are multiple errors in the profiles/printers, for which Cura complains about in the logs and even crashes, throwing an access violation error on startup. This makes it really difficult to identify which one is the culprit of your issue. For example, Cura even sees invalid metadata for some of the definition files that we've had for a long time and we never had problem with!

It could be that something went really wrong somewhere while Cura was saving the configurations or while upgrading them, corrupting everything somehow. But unless we have reproduction steps for that, there is not much we can do to fix the underlying thing that caused this.

@solarstandard, in your case, does the printer show up if you wait for at least 60sec? Because then it might be, as Vagos suggested, the fact that it takes some time to trigger the automatic sync that will show the printer. If that's the case, then @lf-axentia your issue might be a separate one.

solarstandard, 2021-09-27:
My issue seems to be different from what @lf-axentia is having. I want the printer to be added as a cloud printer, but I have LAN access to it as well. Adding the printer via local IP works fine:
[image]

But the _Add a Cloud_ printer dialog is stuck even though the printer is online in the Digital Factory:
[image]

Waiting more than 60 s and the _Check for updates_ workaround don't work either. A cloud connection to the printer is not detected.

Perhaps the reason that I have both a LAN path and a cloud path to the same printer is the problem? E.g. somehow the LAN path overrides the cloud connection attempt? @lf-axentia, can you print to your printers locally?

I should say that I think I have seen the printer icon turn from a check-mark to a cloud icon before. @konskarm, does that indicate whether the connection is LAN or cloud? Unfortunately the icon is a check-mark now, I don't know what happened when it was a cloud.

Btw, I'm doing all of this to make the printer cloud-only. We have have the printer on a medium-sized corporate network and require some form of access control that neither Cura nor the LAN-connected printer itself does not and will not provide. Correct me if I'm wrong here.

Thanks all for keeping this ticket alive. Sorry, for my late response.

lf-axentia, 2021-09-27:
Hi @solarstandard,
I'm able to add my printer both locally and cloud. Cloud shows up automatically after sign in. Local requires me to add printer by IP.
Local connection is indicated witch check-mark.
[image]
Cloud connected by a cloud.
[image]

It is strange that your cloud connected printer doesn't show up. Have you tried to start from a clean configuration as @konskarm recommended? After I tried that all my cloud connect printers was available after sign in.
> Hi @lf-axentia. This is quite a strange issue indeed as we cannot reproduce it locally. The logs don't provide any information that can help us identify the issue either.
> Could you try deleting all [...]

Ghostkeeper, 2021-09-29:
I think the provided [curaconfig.zip](https://github.com/Ultimaker/Cura/files/7217426/curaconfig.zip) from earlier provides a hint as to what could be causing this. There's a weird Sync printer (leftover from when the two printers were a cluster, presumably) which is hidden and connected to the LAN IP address 192.168.41.5. Perhaps it could be that this connection takes over the printer?

I wouldn't know how that mechanism would work to prevent the printer from being shown at all though. But these things are very complex to reason about.
When I load your configuration folder into my Cura it properly shows the S5 printer fine. I can't print to it because I'm not logged in as you. But it does get shown as a cloud printer (just with a greyed out icon).

nallath, 2021-09-29:
Didn't we have an upgrade script that removed those weird configurations? If not; that is an issue that we should solve then.

solarstandard, 2021-09-29:
@lf-axentia, I have deleted the Cura configuration folder contents from Users\[user]\AppData\Roaming\cura\4.11. Starting up Cura again and logging in into the Digital Factory automatically adds my printer, which now has a cloud icon:
[image]
However, going into the Connect via Network dialog the printer shows up with a local IP:
[image]
I'm not sure if that's right.

Also, the Add a Cloud printer dialog still stays stuck. Adding stderr.log file:
[stderr.log](https://github.com/Ultimaker/Cura/files/7252261/stderr.log)

I suppose I should try asking our IT team to bring the printer over into a separate VLAN that can only be accessed via cloud. Perhaps the printer will work with the required cloud-only access then. Please let me know if there are instructions on how to do that properly.

Ghostkeeper, 2021-10-01:
> Didn't we have an upgrade script that removed those weird configurations? If not; that is an issue that we should solve then.

We have, in the upgrade from v4.4 to v4.5. This was done because there was a bug with these hidden container stacks in Cura 4.4. That bug is fixed in 4.5 so that hidden stacks should then be created correctly. We still create new hidden container stacks though.

> However, going into the Connect via Network dialog the printer shows up with a local IP:

We'd expect that the printer then shows up greyed out, under the header that this printer is not controlled via LAN. This might indicate that the printer is not properly connected via the Cloud or somehow has something inconsistent in firmware.

closed 2024-12-09 by github-actions[bot]