Ultimaker/Cura #10470: Cura 4.11 won't launch

cojmeister, 2021-09-17:
### Application Version

4.11.0

### Platform

Windows 10

### Printer

N/A

### Reproduction steps

1. Launched Cura
2. Waited
3. Cura can't start

### Actual results

Recieve a Cura can't start window.
Suggests to backup and reset configuration.
Presents the following traceback error:
Traceback (most recent call last):
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\lib\python3.8\site-packages\cura\Machines\Models\FavoriteMaterialsModel.py", line 11, in __init__
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\lib\python3.8\site-packages\cura\Machines\Models\BaseMaterialsModel.py", line 56, in __init__
TypeError: connect() failed between MachineManager.globalContainerChanged[] and unislot()

With logs:
Thread 0x00002f68 (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.11.0\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 79 in _updateThread
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x000004f4 (most recent call first):
  File "threading.py", line 306 in wait
  File "zeroconf\__init__.py", line 2534 in wait
  File "zeroconf\__init__.py", line 1715 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x000019a8 (most recent call first):
  File "threading.py", line 306 in wait
  File "threading.py", line 558 in wait
  File "C:\Program Files\Ultimaker Cura 4.11.0\plugins\UM3NetworkPrinting\src\Network\ZeroConfClient.py", line 81 in _handleOnServiceChangedRequests
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00001ad8 (most recent call first):
  File "zeroconf\__init__.py", line 1369 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00000e78 (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.11.0\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 61 in _updateThread
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x000004b0 (most recent call first):
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\lib\python3.8\site-packages\UM\Backend\Backend.py", line 165 in _storeStderrToLogThread
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x000004fc (most recent call first):
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\lib\python3.8\site-packages\UM\Backend\Backend.py", line 153 in _storeOutputToLogThread
  File "threading.py", line 870 in run
[... 71 lines cut ...]
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\bin\cura_app.py", line 159 in exceptHook
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\lib\python3.8\site-packages\UM\Qt\Bindings\ControllerProxy.py", line 116 in _onActiveStageChanged
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\lib\python3.8\site-packages\UM\Signal.py", line 332 in __performEmit
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\lib\python3.8\site-packages\UM\Signal.py", line 219 in emit
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\lib\python3.8\site-packages\UM\Controller.py", line 187 in setActiveStage
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 935 in runWithGUI
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 849 in run
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\bin\cura_app.py", line 233 in <module>
  File "Console.py", line 36 in run
  File "__startup__.py", line 74 in run

### Expected results

Cura Launches properly

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

This is weird, because I factory reset my laptop just before this, and I'm having no issues withother programs, neither was I having issues with Cura before the reset
[cura.log](https://github.com/Ultimaker/Cura/files/7186468/cura.log)
.

Ghostkeeper, 2021-09-24:
We've gotten this through Cura's crash reporter as well. Currently 7 reports.
https://sentry.io/share/issue/a8ba916225434d6d8f093819bd14db8b/

Ghostkeeper, 2021-09-24:
On the interwebs I could only find three references to this issue. One was this ticket. The other two:

* https://bugzilla.redhat.com/show_bug.cgi?id=1597386 - They seem to have found a solution, or it solved itself somehow. But I can't figure out what their solution was. People seem to point out that Qt/PyQt and Python may be incompatible? But then why is there a PyQt binary specifically for this Python version?
* https://stackoverflow.com/questions/40674940/why-does-pyqtslot-decorator-cause-typeerror-connect-failed . The accepted answer is that the class is not a QObject, but our class is. I could try to make the called function a `pyqtSlot`, but we connect to this same signal in like 30 different places and none of the other ones are a `pyqtSlot`. It's also strange that it only seems to happen very rarely, or rather, only for some people. We have millions of Windows users after all, and test for Windows thoroughly.

cojmeister, 2021-09-24:
Sadly this is way out of my depth,I'm sorry I can't be of any help.

Ghostkeeper, 2021-10-01:
Sorry, yeah, I wasn't really expecting that of you. I'm just writing down my findings to help other developers who want to investigate this in the future. In short, we've seen crash reports from a couple of people, but most people don't get this crash and we can't reproduce the issue ourselves either.

Something is different on your system than on ours. But we don't know what and don't know what we could ask for to clarify what the difference is. We're currently kind of hoping that someone else comes up with the issue so that we can compare what the similarities are within the computers of the users that get this error, to narrow down what's causing it.

closed 2023-01-29 by cojmeister