Ultimaker/Cura #10471: Hotend crashes into build at the end with print sequence "One at a Time"

gpmfuchs, 2021-09-17:
### Application Version

4.10.0

### Platform

Linux

### Printer

Ender 3v2

### Reproduction steps

- print "One at a Time".
- place a small object in the front right (which has to be printed last)
- place a high object in the center of the build place

### Actual results

- printer prints both objects.
- When the print finishes the hotend is at the front right. (at the height of the small object)
- The End G-Gode moves it to the back left.
- As the Z Axis is not moved high enough, the hotend crashes in the center object.
[image]

### Expected results

The End G-code should move the print head to at least the highest Z level (which exists on the print-bed) (+5mm) before it moves the head.

There should be a variable which holds the highest Z height of all objects on the bed.
So we could start the "End G-Code" with 
G1 Z{highest_Z_object} ; prevent the head from crashing anywhere
G91 ; Relative positioning
...

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[printcrash.zip](https://github.com/Ultimaker/Cura/files/7188456/printcrash.zip)

yet-another-average-joe, 2021-09-20:
This is not a bug, it is the expected behaviour : it's up to the user to arange the models on the bed so there's no collision ; the head is not lifted over parts between prints.
Order should be 1-5-3-2-4 or 1-5-4-3-2.
Print from front to rear, and from left to right or right to left. Not all over the place !

Ghostkeeper, 2021-09-24:
The order cannot currently be determined by the user (see https://github.com/Ultimaker/Cura/issues/4835). Cura should order the objects so that the print head doesn't collide with neighbouring objects while printing. However it doesn't consider the travel move between different objects in that. That's not a problem, because CuraEngine is expected to go to the height of the highest model printed so far, plus 5mm clearance:

https://github.com/Ultimaker/CuraEngine/blob/f460e01cd154fb72663f1e1e3cd2c3a960d57191/src/FffGcodeWriter.cpp#L147

Noteworthy is that `max_model_height` is updated as we go here:

https://github.com/Ultimaker/CuraEngine/blob/f460e01cd154fb72663f1e1e3cd2c3a960d57191/src/FffGcodeWriter.cpp#L147

This function is called separately for every mesh group (one-at-a-time printed group of models). So the `max_model_height` grows monotonically when taller objects are encountered. However this shouldn't be a problem, because you can only collide with objects after they have been printed.

If you look carefully at the g-code though, you'll find that the travel moves between objects are actually fine.
[image]
It prints these three from left to right. The travel move from the first (low) object to the second (tall) object is enough to clear the low object. The travel move from the second (tall) object to the third (low) object is tall enough to clear the tall object. The problem is the final travel move, which is introduced because of the end g-code of this printer:

```
G91 ;Relative positioning
G1 E-2 F2700 ;Retract a bit
G1 E-2 Z0.2 F2400 ;Retract and raise Z
G1 X5 Y5 F3000 ;Wipe out
G1 Z10 ;Raise Z more
G90 ;Absolute positioning

G1 X0 Y{machine_depth} ;Present print
M106 S0 ;Turn-off fan
M104 S0 ;Turn-off hotend
M140 S0 ;Turn-off bed

M84 X Y E ;Disable all steppers but Z
```

This "raise Z more" line moves it up 10mm which is not enough to clear all the models. And then it moves to `X0 Y220` blindly.

A simple solution would be to just move to `G1 Z{machine_height}`. That's effectively what the Ultimaker printer does in firmware when the print finishes. It's a little less efficient than something like `G1 Z{max_model_height}`, but are those 5 seconds really going to matter much?

yet-another-average-joe, 2021-09-29:
Thanks for the explanations ! I realize thst I have been tricking Cura since "One at a Time" was introduced (4.0 ?), and completely forgot about it. My printer parameters are set so the head dimensions are not taken into acount, and manually move the models so they can be printed. (by eye, knowing my printer, and using the layers slider control)

Moreover, I didn't 100% understand the question (small vs large instead of small vs tall)...

Having a look to this project file, you will understand how I'm using "One at a Time"

[OneAtATime.zip](https://github.com/Ultimaker/Cura/files/7254908/OneAtATime.zip)

Sorry for the out of topic answer. Anyway, I think that this cheat can be helpfull in this context.
Will probably not work with a Ultimaker printer, depending on models heights !

gpmfuchs, 2021-10-01:
@Ghostkeeper 
It takes quite a while for the Ender3v2 to get to its highest position.
But there are 2 reasons I don't want to use this solution:
1. the bowden tube bends on the top beam if Z ist near its max, which I prefer only happens when necessary.
2. The start gcode starts with "move up 1cm" (to prevent the hotend from destroying the bed and to not rub filament from the last print onto the bed, until z height is established)
So I would reach my physical Z limit whenever I start a new print.

GregValiant, 2024-11-16:
This is still a problem in Cura 5.9.0-beta.2.  The nozzle will leave the model from the current Z height and then if the Ending Gcode has a park position in it, the nozzle can crash into a taller model.
There should be a final Z move up to the transit height like there is when a move is from "this model ending" to "next model start".
;TIME_ELAPSED:3758.542064
G1 F2100 E705.88126
G0 F2100 Z24.5               ; one_at_a_time final Z move
M140 S0
M205 X30 Y30
M107

GregValiant, 2024-11-17:
@HellAholic ,
In machine design; special care was always given to "Show all interferences".  (We always laughed at that because if there was truly an interference, we would have changed the design.)
It did make sure we thought about checking for any crashes.  We _HATED_ crashes.  They were expensive.

I know just enough about this to be dangerous.  Ignoring my syntax (which is likely wrong) Would something like this work?
From ["FffGcodeWriter.cpp" line 4403](https://github.com/Ultimaker/CuraEngine/blob/7876d035c30d17a0f0e48568913e7d47ce4a3223/src/FffGcodeWriter.cpp#L4403):
The intent would be to add a line:  "G0 F{speed_z_hop} Z{max_object_height + 5mm} ; Final Z Move" which would place the nozzle above the top of the tallest print and insure any travel moves that might be in the Ending Gcode would not crash into a print on the build plate.

```
void FffGcodeWriter::finalize()
{
    const Settings& mesh_group_settings = Application::getInstance().current_slice_->scene.current_mesh_group->settings;
    // If print_sequence is 'One-at-a-Time' then add a Z move up to the transit height.
    if (mesh_group_settings.get<str>("print_sequence") == "one_at_a_time")
    {
        gcode.writeTravel(Point3LL(p.X, p.Y, max_object_height + MM2INT(5)), speed);; // Final Z move.
    }	
    if (mesh_group_settings.get<bool>("machine_heated_bed"))
    {
        gcode.writeBedTemperatureCommand(0); // Cool down the bed (M140).
        // Nozzles are cooled down automatically after the last time they are used (which might be earlier than the end of the print).
    }
    if (mesh_group_settings.get<bool>("machine_heated_build_volume") && mesh_group_settings.get<Temperature>("build_volume_temperature") != 0)
    {
        gcode.writeBuildVolumeTemperatureCommand(0); // Cool down the build volume.
    }
```