Ultimaker/Cura #10472: Cura has a print bed calculation error with multiple extruder offsets

Lumanet2012, 2021-09-17:
### Application Version

4.11.0

### Platform

Windows 10

### Printer

Heavy modified Creality Ender 3 

### Reproduction steps

Hello,

I have a DUAL Extruder setup where my center point is my Bed probe at 0,0,0 and 
Extruder 1 has an offset to the left so it is X = -34.5mm      Y = 1mm 
Extruder 2 has an offset to the right so it is X = 34.5mm     Y = 0 mm 

Extruder 1 is only used for support at this point. 
Extruder 2 is used for the print filament.   

Here are screenshots of the setup : 
https://owncloud.lumanet.info/s/XpQic8oQJ2e4MfG 

### Actual results

For some reason or the other CURA is not doing it's math correctly. 

As you can see in the photos attached : 
https://owncloud.lumanet.info/s/XpQic8oQJ2e4MfG

Cura calculate and NO PRINT zone of twice the offset value ignoring that both are offset in different X direction. 

So I would understand that CURA set a no print zone for 34.5mm on either side, but not 70+mm on each side. 

Screenshot_88.png shows the right side opens up for printing when I disable the usage of Extruder 1 which is holding my support material. 

### Expected results

Complex printers do need Complex settings : 

For example tool head parking and tool whip function are completely ignored by CURA.

My X gantry can move over at BLTouch (X0, Y0) much more than just 235mm. 

What I need to see is more values to be added within CURA printer settings : 

XA = Left moving area 
X  = build plate width 
XB = Right moving area 

This way I would be able to see my real PRINT area been reflected without been cut out in either way.

Same should apply if someone has print heads running infront of each other like with an Y offset.  

ONLY if those settings are proper added we will than see a proper print area as well. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Screenshots from my current setup are to be found here : 

https://owncloud.lumanet.info/s/XpQic8oQJ2e4MfG

Picture  Photo-2021-08-18-12-50-28_9724.JPG  shows the actually full setup of the print head.

BLTouch is the center. 

My X Gantry has actual a movement of over 300mm as well but because of the Extruder offset's of ~35 on either side the print head can move and push out each extruder over the actual print bed size where actually is a WIPE area. 

So X_Min causes the nozzle wipe on for Extruder 1  and X_Max causes a nozzle wipe for Extruder 2. So on both sides of the bed filament is wiped off before been printed.

Ghostkeeper, 2021-09-18:
> Extruder 1 has an offset to the left so it is X = -34.5mm Y = 1mm
Extruder 2 has an offset to the right so it is X = 34.5mm Y = 0 mm

So, to make sure we understand each other, this means that there is a space between these two nozzles of 69mm, right?

So that means that there is a band of 69mm wide on the left side where the right nozzle can't reach. And a band of 69mm on the right side where the left nozzle can't reach, right? Like, if the left nozzle is all the way to the left as far as the print head will go, then the right nozzle is still 69mm away from the left side.

So if only extruder 1 is used in the print, it would have a space on the right side of the build volume where that extruder can't reach, with a width of 69mm. And if extruder 2 is used in the print, it would have an equally wide space on the left side of the build volume. If both extruders are used in the print, there is a space on both side where one of them can't reach and you'd have a very limited build volume.

Cura will disallow printing outside of the build volume. So the build volume in Cura is the union across all possible places that either extruder could reach. This will be made smaller by your 69mm spacing between the nozzles. Doesn't matter if that spacing is -60 and +9, or -34.5 and +34.5. It's the same physical distance.

Lumanet2012, 2021-09-18:
> 
> 
> > Extruder 1 has an offset to the left so it is X = -34.5mm Y = 1mm
> > Extruder 2 has an offset to the right so it is X = 34.5mm Y = 0 mm
> 
> So, to make sure we understand each other, this means that there is a space between these two nozzles of 69mm, right?
> 
> So that means that there is a band of 69mm wide on the left side where the right nozzle can't reach. And a band of 69mm on the right side where the left nozzle can't reach, right? Like, if the left nozzle is all the way to the left as far as the print head will go, then the right nozzle is still 69mm away from the left side. This is why the border has a width of 69mm in your case.
> 
> So if only extruder 1 is used in the print, it would have a space on the right side of the build volume where that extruder can't reach, with a width of 69mm. And if extruder 2 is used in the print, it would have an equally wide space on the left side of the build volume. If both extruders are used in the print, there is a space on both side where one of them can't reach and you'd have a very limited build volume.
> 
> Cura will disallow printing outside of the build volume. So the build volume in Cura is the union across all possible places that either extruder could reach. This will be made smaller by your 69mm spacing between the nozzles. Doesn't matter if that spacing is -60 and +9, or -34.5 and +34.5. It's the same physical distance.
> 
> > My X Gantry has actual a movement of over 300mm as well but because of the Extruder offset's of ~35 on either side the print head can move and push out each extruder over the actual print bed size where actually is a WIPE area.
> 
> Two cases are easy to configure Cura for: Either the build volume is limited by the range of the gantry, or it is limited by the size of the build plate only, assuming infinite gantry range. You can choose between these with the `nozzle_offsetting_for_disallowed_areas` metadata on the printer definition (not available from the GUI). For other cases, where the limit is the build plate in some cases but the limit is the gantry in other cases, you have the `machine_disallowed_areas` setting and the `nozzle_disallowed_areas` setting. These allow for arbitrary polygonal disallowed areas that move or don't move with the nozzle offset. I think you would have to write that wipe area down in a disallowed area polygon for your case.

Ghostkeeper I think you should recall your statement because you don't make any sense.  
Yes between the 2 nozzles is a distance of 69mm which you can clearly see in the photos I had attached to this BUG. 
Again you missing the POINT that each extruder is offset of the BLTOUCH which is  the Center. Which again set's it to 34.5mm 
and not as CURA shows 69mm.

nallath, 2021-09-20:
> Ghostkeeper I think you should recall your statement because you don't make any sense.

I don't think that not having consensus about the exact problem is a reason to recall / retract statements. As far as I understand it, I think Ghostkeepers description is correct.

Lumanet2012, 2021-09-22:
> 
> 
> > Ghostkeeper I think you should recall your statement because you don't make any sense.
> 
> I don't think that not having consensus about the exact problem is a reason to recall / retract statements. As far as I understand it, I think Ghostkeepers description is correct.
> 
> If you think we're somehow wrong, or how our interpretation isn't right, could you please try to explain it differently?

Nallath. Did you even looked at the photos which are attached to this ?  

BLTouch is 0,0  not any of the nozzles. So when you push the print head to the left for X0  the BLTouch only is 34mm off to the side of the bed.  Meaning The nozzle of extruder 1 which is the one on the right is barely touching the glass of the print bed. 
same is when you move the print head all the way to the right. 
Similar setup as the IDEX but with the difference it's a single carriage unit.  So blocking off a printing area on the printbed of 69mm so 6.9 cm is simply wrong because the nozzles both can reach the X235mm print area. 

Again what Cura is still not getting right is when print bed's do have a OFFSET in X&Y of the real zero.  Which is an ongoing issue with Cura because every other slicer stepped up and allowed offset's been set up for X&Y where Cura still thinks every printer has ZERO ZERO close the the corner of the print bed.

Ghostkeeper, 2021-09-24:
Cura doesn't know where your 3DTouch is located, and doesn't care. Where the 3DTouch is located doesn't matter for the range where your nozzles can print. It's irrelevant here. What matters is which firmware coordinates map to which physical coordinates, and the offsets between the nozzles.

> Again what Cura is still not getting right is when print bed's do have a OFFSET in X&Y of the real zero. Which is an ongoing issue with Cura because every other slicer stepped up and allowed [...]

I think this is your real problem here. Cura has the origin either in the front-left, or in the centre of the build plate. Not in any other place.

This is easily solved though. Just put this g-code in the start g-code:

```
G0 X0
G92 X34.5
```

This tells the printer to move to a known position (`X0`), which in your case would still be 34.5mm away from the limit of the left nozzle position, and then tells it that this is actually position `X34.5`. Effectively this aligns the minimum position of the nozzle to X0.

Indeed, Cura doesn't have an easy text field for all cases of all printers out on the market. Having 0,0 be the farthest point that the nozzles can reach is common due to the nozzle switches usually being mounted to hit the print head, not the nozzles. Some printers have a different way to set the origin position or do offsets in their firmware. That's what the start g-code field is meant to handle though: All special cases where the easy-use GUI elements don't provide a sufficient solution.
Similarly, the start g-code may also need to be modified in order to prime correctly, or do bed levelling correctly, or associate the correct fans with the correct extruders, or set the correct number of steps per mm, enable motor power, and a myriad of other things. Cura's default options work for almost all of the 430 printers it has profiles for. The three MakeIt Pro printers are the only ones among those that need this solution as well because their build plate is 6mm offset in X and Y. For uncommon situations such as that, adding two lines of start g-code is fine.

Let me know if adding these lines to the start g-code (and subsequently adjusting the build volume size) works for your printer!