Ultimaker/Cura #10474: Wall

boduk-koseki, 2021-09-18:
### Application Version

4.11.0

### Platform

win10

### Printer

SW-X1 V4

### Reproduction steps

Setting Visibility

Z seam position-> ON / OFF parameter check does not reflect in profile tree

Relative Z seam> ON / OFF parameter check does not reflect in profile tree

### Actual results

Restart CURA 4.11.0

### Expected results

No change after restarting CURA 4.11.0

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

no

nallath, 2021-09-20:
I don't understand what you're issue is. Your actual result is that Cura restarts? Do you mean that it crashed? If so, we would need logs to find out what happened.

Ghostkeeper, 2021-09-24:
Note that the Z Seam Position and Relative Z Seam settings are only visible if the Z Seam Alignment setting is set to "user-specified". It should show a warning in the setting visibility screen about this.

closed 2021-10-04 by no-response[bot]