Ultimaker/Cura #10480: Can´t 3D Print anymore

Stalk3rplay, 2021-09-19:
### Application Version

4.11.0

### Platform

Windows 10

### Printer

Anycubic I3 Mega S

### Reproduction steps

Before Update to 4.11.0. my printer worked normaly. Since update, cura opens my files as "AI3M_ ..." and after slicing, saving to sd card and start printing, my printer does nothing except making weird noises. Nothing changed in Autodesk Fusion 360, no update, nothing changed on my printer, no update. Opening old files doesn´t work either anymore. I downgraded to the last version of Cura i succesfully worked with (4.9.1.), still opens files as "AI3M_ ...". Before all that, Cura opened files as "CFFFP_ ..." and my printer printed without any problems.

New project:
[asus boden 2.zip](https://github.com/Ultimaker/Cura/files/7191984/asus.boden.2.zip)

Old project (worked before the issue):
[AudiTTRoadster.zip](https://github.com/Ultimaker/Cura/files/7191985/AudiTTRoadster.zip)

Printer file that worked before:
[CFFFP_Spiegel v1.zip](https://github.com/Ultimaker/Cura/files/7191989/CFFFP_Spiegel.v1.zip)

Printer file that doesnt work anymore:
[AI3M_asus boden rechts.zip](https://github.com/Ultimaker/Cura/files/7191991/AI3M_asus.boden.rechts.zip)
 

### Actual results

Nothing changed, I tried exporting other file types my printer supports from Autodesk Fusion 360, still the same problem. 

### Expected results

Cura opening files as "CFFFP_ ...", Printer working fine again.
Or help to find where the problem could be else?

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Printer tested with old files from sd card -> Works fine
-II- with new type of file that cura creates -> Printer starts progress, doesn´t print, only making weird noises
Reinstalled Autodesk fusion 360
Reinstalled Cura 4.11.0
Downgraded to Cura 4.9.1 where the problem doesn´t existed
Tried software update on printer
Still the same problem

GreyExe, 2021-09-19:
I have a similar problem. I cannot generate GCODE's that consist of one layer (bed calibration test). The GCODE only consists of a start and end code. This works perfectly with Cura 4.10.

nallath, 2021-09-20:
> I have a similar problem. I cannot generate GCODE's that consist of one layer (bed calibration test). The GCODE only consists of a start and end code. This works perfectly with Cura 4.10.

That is a a completely separate issue: https://github.com/Ultimaker/Cura/issues/10394

nallath, 2021-09-20:
Uh. It seems to be related to what printer you selected in Cura (a custom ff printer vs the anycubic i3). Unfortunately, I don't know for sure, since you didn't upload project files, only stl files.

So if you could upload 3MF files, we could check if that is indeed the case.

Jman64cola, 2021-09-20:
I noticed the same this has happened to my printer, i had printed something on cura 4.10 with no problems, then i installed cura 4.11 and i have 4 different prints. each of them will not print on my printer. i noticed they do not set the temp or wait for it to hit a level. it just sit there. i run an ender 3 v2 with Jyers new firmware on the 4.2.2 board with manual bed leveling.

nallath, 2021-09-21:
> I noticed the same this has happened to my printer, i had printed something on cura 4.10 with no problems, then i installed cura 4.11 and i have 4 different prints. each of them will not print on my [...]

Okay, could you then provide the info that i requested? I simply don't have enough to debug it right now.

Jman64cola, 2021-09-21:
I have attached the 3mf file for the prints that failed to start. 
i have also attached the gcode of 2 prints i was able to start with no problems and the other code that will not print. i have tried 4 different prints and since using cura 4.11 it will not print. also going back to 4.10 cura does not work either now. 
i have label these items accordingly: 
[print worked (2).txt](https://github.com/Ultimaker/Cura/files/7203209/print.worked.2.txt)
[print worked.txt](https://github.com/Ultimaker/Cura/files/7203210/print.worked.txt)
[print failed to start.txt](https://github.com/Ultimaker/Cura/files/7203211/print.failed.to.start.txt)

https://drive.google.com/file/d/1tKqmAUodS1JIJ6H_eFn-XJRXAz-Hu9Qs/view?usp=sharing

Jman64cola, 2021-09-21:
also the print worked.txt and print failed to start.txt are the same model just generated the g code at different times on different versions of cura.

Jman64cola, 2021-09-21:
https://drive.google.com/file/d/1tKqmAUodS1JIJ6H_eFn-XJRXAz-Hu9Qs/view?usp=sharing 

i saved this as a project and re uploaded it. let me know if that works

GregValiant, 2021-09-21:
It's something @Jman64cola because the "Print Failed to Start" file printed fine on my Ender 3 Pro.  So did @Stalk3rplay's "AI3M_asus boden rechts.gcode" file.  There are a couple of oddities in that AI3M file though.
G1 F50
G0 Y20 F50
M117 Printing...
G5
A speed of .83mm/sec is odd as is the G5 "Bezier Curve" command.

nallath, 2021-09-21:
Maybe the arc welder is enabled?

fieldOfView, 2021-09-21:
The Arcwelder plugin only makes arcs (G2/G3), not beziers (G5).

Ghostkeeper, 2021-09-27:
Could you please write down reproduction steps as a numbered list of steps? From this story, I don't know what I should try to reproduce the issue, nor what you would expect to happen once those steps are executed.

As I understand it now, you select a specific printer (Anycubic I3 Mega? Or Custom FFF?), you save the g-code (via your phone) to an SD card, and then try to select that file for printing on the printer. And what happens then? Does the printer sing a song? Are the motors going very slowly? Is the print head running into the side of the printer?

closed 2021-10-04 by no-response[bot]