Ultimaker/Cura #10481: Show time difference / change from previous slice

kthornbloom, 2021-09-19:
### Is your feature request related to a problem?

After (hopefully) optimizing a bunch of settings, I re-slice to see if it's faster. Sometimes I forgot to note the previous time, and I'm not sure if the new settings are better or worse.

### Describe the solution you'd like

Each slice, show a time difference with the previous slice:

[image]

### Describe alternatives you've considered

I've tried remembering the old time estimate, but my brain no work good.

### Affected users and/or printers

Errybody.

Vandresc, 2021-09-20:
Hi @kthornbloom , thank you for your feedback, I will bring your request up to the team.

Vandresc, 2021-09-23:
Devs for our own reference, see CURA-8583 in our backlog.

fvrmr, 2022-03-21:
We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs. 

To be able to focus on the most requested and needed features from our community and from Ultimaker, we have decided to remove this feature request from our backlog. We believe we will not work on this anytime soon and will therefore defer this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this feature and create a pull request yourself.

kthornbloom, 2023-06-29:
Yes, this is absolutely a feature I would still like to see happen. I think it would be extremely easy to implement (just some addition/subtraction here), and would be very useful to see at a glance what you've been able to optimize. 

Here's a fresh UI demo:

[image]

MariMakes, 2023-08-02:
Quick update from our side 👋

Thanks for upvoting this issue again. I showed it to the team and they are still excited to work on this, we have new insights that might make implementation of this easier. We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-10897

Thanks for the report! 👍

kthornbloom, 2023-08-03:
Thanks for the update! Looking forward to it