Ultimaker/Cura #10482: Exporting produces garbage files

Kaelum, 2021-09-19:
### Application Version

4.11.0

### Platform

Windows 10

### Printer

All (Creality Ender 5 Plus)

### Reproduction steps

1. Create a custom Profile
2. Export the Profile
3. Open the exported Profile in a text editor.
4. The file is partially garbled and can't be imported.

### Actual results

Garbled files.

### Expected results

The full Profile in a readable/importable format.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

This has been broken for well over a year.  I don't remember the first release that broke it, but it is impossible to keep systems in sync now.

smartavionics, 2021-09-19:
It's a zip file.

nallath, 2021-09-20:
As SmartAvionics already mentioned; it's not a text file. It's a zip file that contains multiple text files. 

This isn't broken and this has been the case since Cura 2.1

closed 2021-09-20 by nallath