Ultimaker/Cura #10483: trackpad rotation not working

farm24, 2021-09-19:
### Application Version

4.11.0

### Platform

windows 10

### Printer

custom 

### Reproduction steps

1. start up cura
2. attempt to rotate the camera by holding right click and dragging

### Actual results

the camera doesn't rotate

### Expected results

the camera should rotate

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

i cant find anything online to solve this issue but hold down space and do it but that is incredibly uncomfortable on my laptop

Vandresc, 2021-09-20:
Hi @farm24 , that is indeed the current behaviour when using the touchpad. Space bar should be pressed when rotating the view. 
When using a mouse then right click and dragging the mouse will rotate the view. 

As this is how we implemented I will consider your feedback a feature request and will bring it up with the team.

Thank you for your feedback.

farm24, 2021-09-20:
Oh ok I got confused because the Mac version you just have to right click and drag

evtrados, 2021-09-22:
Hi @farm24 ,

I checked this both using Windows 10 (on a desktop PC, an HP laptop and a Lenovo laptop) and macOS Catalina and I see three ways that you can rotate:
1. Using right click and dragging
2. Using left click + Space and dragging
3. Using right click + Space and dragging

On macOS there is no right click button on the trackpad, so that means that you'll of course need to do the two-finger click on its right side in order for the OS to identify it as a right mouse click.

Could it be that the drivers of your trackpad need to be updated on your windows machine? 

For now, we cannot reproduce the issue within the team, so we will unfortunately have to defer it.

ryanalexmartin, 2023-01-11:
@Ghostkeeper re-opened this issue, but it was marked as completed.  The original issue was marked as completed.

I believe the issue should be closed, and if @Ghostkeeper encountered an issue with right-clicking not working, it should be created as a new issue, with relevant details and a more detailed bug report.

dzid26, 2023-05-17:
> Using left click + Space and dragging

Except it only works if the touchpad is hard pressed (with an audible switch click). That is _Double-click and hold to drag_ is still zooming, instead of rotating. This is unlike to `Shift` modifier that _does_ work with _Double-click and hold to drag_ which makes you think the `SPACE` is _not_ working. 

On top of that `SPACE` modifier only works sometimes. 
And another times, after certain maneuvers like alt+tab, Cura thinks the left button is still pressed and I need to restart it.

Now I noticed that if I open `.gcode` with Cura, the `SPACE` modifier doesn't work if I first close this warning:
[image]
And if don't close it and start maneuvering, the left button is stuck and I cannot stop zooming. 

The `SPACE` is a terrible choice of modifier anyway and I want `Ctrl` or `Alt` like other programs.