Ultimaker/Cura #10485: Cura crashes when I add the Under 3 printer / New and clean installation | Cura plante quand j'ajoute l'imprimante Under 3 / Installation neuve et propre

ThomasTheFrench, 2021-09-20:
### Application Version

4.11.0

### Platform

Win 10 pro 21H1 build 19043.1237 / Sys x64 / Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz   2.21 GHz / 16G0 RAM

### Printer

Creality Ender 3

### Reproduction steps

Note : I'm French, sorry for my poor English...

I freshly install Cura 4.11.0, and when I try to install the Creality Ender 3 printer, Cura show me the settings panel and the screen freeze and crash and disappear. I try with other printer like Utimaker printers  and I don't uncounter this issue... I had before Cura 4.9 but I uninstal it complettly and I cleaned my computer. I try also to remove the %APPDATA% files but it don't change anything.

------

J'ai récemment installé Cura 4.11.0, et lorsque j'essaie d'installer l'imprimante Creality Ender 3, Cura m'affiche le panneau des paramètres et l'écran se fige et disparaît. J'essaie avec d'autres imprimantes comme les imprimantes Utimaker et je ne rencontre pas ce problème... J'avais avant Cura 4.9 mais je l'ai désinstallé complètement et j'ai nettoyé mon ordinateur. J'essaye aussi de supprimer les fichiers %APPDATA% mais cela ne change rien.

Log :

[stderr.log](https://github.com/Ultimaker/Cura/files/7195994/stderr.log)
[cura.log](https://github.com/Ultimaker/Cura/files/7195993/cura.log)

Last screen befor the crash and clossing of Cura :
Dernier écrans avant le crash de Cura :

[image]

### Actual results

Crash when I try to add Creality Ender 3 printer

### Expected results

Add Creality Ender 3 printer successfully and use normally Cura.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

-

nallath, 2021-09-20:
This is caused by this upstream issue: https://github.com/scipy/scipy/issues/8850

ThomasTheFrench, 2021-09-20:
The bug exist because I am using in the path of the folder "temp" an "non-english" character like this issue : [https://github.com/scipy/scipy/issues/8850](url).

THE SOLUTION : 

Change the environnement varible who cantains the paths of the folder "temp" by a new path without "non-english" characters.

Exemple : 

The wrong path for the folder "temp" is :  C : /users/%%your username%%/......./TEMP with an "non-english" chararcter in the username.

The new path is now : D:/win_temp with only "english characters" (you can choose the location you want)

Note : restart your computer after change.

I would like to draw the developer's attention to this bug: it's a crap bug that shouldn't exist ... I'm comfortable with computers but imagine the others that aren't ... It's not just English speakers who use this software so many people have to use "non-English" characters in their windows username. So this case must happen often. It would be nice to fix this bug.

-----

Le bug existe car j'utilise dans le chemin du dossier "temp" un caractère "non anglais" comme dans ce bug: [https://github.com/scipy/scipy/issues/8850](url).

LA SOLUTION :

Changez la variable d'environnement qui contient les chemins du dossier "temp" par un nouveau chemin sans caractères "non anglais".

Exemple :

Le mauvais chemin pour le dossier "temp" est : C : /users/%%your username%%/......./TEMP avec un caractère "non anglais" dans le nom d'utilisateur.

Le nouveau chemin est maintenant : D:/win_temp avec uniquement des "caractères anglais" (vous pouvez choisir l'emplacement que vous voulez)

Remarque : redémarrez votre ordinateur après modification.

Je voudrais attirer l'attention du développeur sur ce bug : c'est un bug de merde qui ne devrait pas exister... Je suis à l'aise avec les ordinateurs mais imaginez les autres qui ne le sont pas... Il n'y a pas que les anglophones qui utilisent ce logiciel tant de gens doivent utiliser des caractères "non anglais" dans leur nom d'utilisateur Windows. Donc ce cas doit arriver souvent. Ce serait bien de corriger ce bug.

closed 2021-09-20 by ThomasTheFrench

Ghostkeeper, 2021-09-27:
See also https://github.com/Ultimaker/Cura/issues/9644 for Cura's own tracking of this bug, although it is considered an upstream bug.