Ultimaker/Cura #10487: Support blocker options not saved

woonex, 2021-09-20:
### Is your feature request related to a problem?

When adding support blocker objects into the model and using them to modify slicing parameters, it always reverts to a default set of parameters. 

### Describe the solution you'd like

It would be nice if the support blocker modify settings for overlaps would allow you to customize it similar to the settings menu on the right. It appears to use the same interface selection method as the advanced settings menu under the profile, so I'm not sure why it can't save the user selections. 

### Describe alternatives you've considered

No alternatives have been considered

### Affected users and/or printers

Everyone should be able to see the value. There have been some times I've had to have 20+ support blocker objects where I have to go into the settings menu and search for the settings I need to modify for each instance. 

### Additional information & file uploads

[image]

nallath, 2021-09-20:
I think this is probably a duplicate of this issue: https://github.com/Ultimaker/Cura/issues/8776

woonex, 2021-09-20:
It's similar to that, but it's mainly the issue that the options window for the modifier mesh options ALWAYS reverts to a default set of modifying options. Starting a new project will require selecting options for infill density, print speed, etc. as will using a new object for a modifier mesh.

Ghostkeeper, 2021-09-27:
This is a request to add a profile system to the per-object settings. Indeed, currently the per-object settings can't be saved to a profile and you can't load those settings back in except through a project file (but then you'll have the same model).

I consider this to be a duplicate of https://github.com/Ultimaker/Cura/issues/3192.

closed 2021-09-27 by Ghostkeeper