Ultimaker/Cura #10489: Seperate line width for top and bottom layers

DragonFlyer007, 2021-09-20:
### Is your feature request related to a problem?

Often only one side of the printed model needs a fine finish, and a smaller line width on that layer can absolutly help with that. It will give that layer a higher "resolution".

Cura can only set both top and bottom the same.

### Describe the solution you'd like

Individual line width settings for top and bottom layers.

### Describe alternatives you've considered

None

### Affected users and/or printers

This will benefit everyone much!

Ghostkeeper, 2021-09-27:
Currently the top and bottom skin areas are joined together into one area. This prevents getting a separation between those areas like what Top Surface Skin does:
[image]

Introducing different pattern settings for the top vs. bottom skin will add that border. This will be visible in the print.

DragonFlyer007, 2021-09-29:
> Currently the top and bottom skin areas are joined together into one area. This prevents getting a separation between those areas like what Top Surface Skin does: [image]
> 
> Introducing different pattern settings for the top vs. bottom skin will add that border. This will be visible in the print.

But how can you change pattern for bottom skin?  The line direction setting? SEt to "[0,90]" now, But how could you change the bottom skin layer here?

Thanks

closed 2021-10-06 by Ghostkeeper