Ultimaker/Cura #10493: Pas d'impression correct

YAPRINT37, 2021-09-21:
### Application Version

4.11

### Platform

Windows 10

### Printer

CREALITY ENDER 3

### Reproduction steps

J'ai donc programmer mon impression, fait le découpage, enregistrer mon impression sur la clef et pour finir j'ai lancé l'impression

### Actual results

Les differents essais que j'ai pue effectuer son revenu a une impression qui et louper c'est a dire:

> L'IMPRESSION fait bien sont trait sur le plateau pour ammorcer l'extrudeur, aprés il vas commencer à imprimer, la buse se trouvant a 1 centimètres au dessus du plateau a ce moment la l'impression ne peut pas ètre bonne, il déssine bien les formes de l'élément mais comme il se trouve à 1 cm au dessus du plateau cela ne génère aucune forme.

>  L'impression commence normalement, jusqu'a une certaine hauteur ou il doit aller trop haut sur l'axe Z donc il laisse sortir le filaments sans le lisser avec la buse comme il et trop haut .

### Expected results

L'impression ne devrais pas beuger comme ceci en plein millieu ou au debut.
Sachant que j'ai un logiciel de secours que j'ai essayer pour voir si cela venait pas de l'imprimante et avec ce logiciel tout fonctionne correctement.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

je fournirais les photos sur demande

YAPRINT37, 2021-09-21:
The various tests I was able to perform his tempering to an impression that is missed that is: the IMPRESSION does well are strokes on the tray to prime the extruder, after it will start printing, the nozzle is located 1 centimeter above the tray at this time-there the impression can not be good, it draws well the forms of the element but as it is 1 cm above the tray it does not contain any form.

[Projet.zip](https://github.com/Ultimaker/Cura/files/7203868/Projet.zip)

I can’t provide you with the print log because I can’t find it

YAPRINT37, 2021-09-21:
In case of troubleshooting I have a second software, which to try and that to work. So I think of a problem other than the printer

Ghostkeeper, 2021-09-28:
If we are to help with this issue, we need to know what g-code Cura needs to output differently in order to make it work for your printer.

I don't have your printer. I can't see what's going wrong at your place. It could be a broken Z stepper, or maybe automatic bed levelling or some firmware bug. All I can see is what Cura outputs, and from what I can see that looks to be correct g-code flat on the build plate.