Ultimaker/Cura #10494: Part Select and Move Tools Non-Functional (macOS + Fusion 360 Export)

BrandonParod, 2021-09-21:
### Application Version

4.11.0

### Platform

macOS v11.5.2 (2.6GHz 6-Core Intel Core i7

### Printer

Ultimaker S5

### Reproduction steps

1. open cura
2. export mesh from Autodesk Fusion 360 using Save As Mesh command
[image]
3. model is imported and placed on build bed in Cura
[image]

### Actual results

I cannot click on the parts to select them, click on the gizmo arrows to move the part, or use any of the other tools. The parts can only be selected from the object list but the behavior is the same as listed above.

### Expected results

The parts should be clickable and all of the tools should work

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7205623/cura.log
[Part Select Bug.3mf.zip](https://github.com/Ultimaker/Cura/files/7205633/Part.Select.Bug.3mf.zip)
)

BrandonParod, 2021-09-21:
After some more searching it looks like this is related to the closed issue #8745. One of the comments on that issue mentioned that moving the window to the laptop screen (I have two external monitors) allowed parts to be selected again. This worked for me. Anything advice on what I can do to restore functionality on my main monitors would be greatly appreciated!

Ghostkeeper, 2021-09-28:
Does this also happen when you plain import an STL file rather than via Fusion? I don't expect that makes any difference really.

Also weird that the tool handle arrows appear that big, but not the rest of the models! I've never seen anything like that!

BrandonParod, 2021-09-30:
Thanks for the follow up! STL file import makes no difference. This has been persisting despite restarting computer, fusion, Cura, and disconnecting/reconnecting the external monitors (running off of TBT3 dock). I can only get it to work properly when running on the MacBook Pro display.

fieldOfView, 2021-10-07:
@evtrados do the internal and external displays you tested have different DPIs? So a retina and a non-retina display?

PeterSawyer, 2021-11-15:
This is still happening on 4.12.0 macOS Monterey 12.0.1

My workaround is to click the green button to make Cura enter fullscreen, you can then either work on it in full screen or exit full screen and it will work.  Moving between my displays did not fix anything.