Ultimaker/Cura #10495: Z axis stepping error

dn7050, 2021-09-21:
### Application Version

4.11.0

### Platform

Windows 10

### Printer

Ender 3 Pro, SKR 1.4, Microswiss direct extr, BLT

### Reproduction steps

Recently calibrated  X, Y and Z axis motion.  All, including Z axis, were tested and adjusted over 200 mm travel and now tests exact at 100 and 200 mm travel.  Note that BLTouch is used for Z homing and the nozzle offset is -2.3mm.  This produces a good first layer.

### Actual results

Printing a standard 20mm test cube results in accurate X and Y dimensions but Z is about 22.3 mm.   Enlarging the cube to 40mm results in accurate X and Y but Z is now 42.3mm.  This result was repeated over printing several calibration cubes. Doing travel tests measuring Z travel from "0" (Nozzle at 0.1mm above bed) to 10mm using the Z motion control, results in an actual travel of 12.3 mm even though the display shows a travel of 10mm.  Inputting an additional 10mm gives a measured additional travel of 10 mm.  Stepping up Z axis from "0" 1mm at a time shows that the first 10 mm or so is actually moving up more that 1 mm at a step so that the total by 10 mm is about 12.3mm when only ten 1mm steps were input.  Then above 10mm it stabilizes to 1mm steps as it moves higher from the bed.  It appears to be somehow including the -2.3mm offset in the first 10 mm of commanded travel by adding 2.3/10 to each one mm commanded up to the 10mm level.  How can this be corrected?  Note that this does not show in the slicer.  After slicing using 0.16, 0.2 and 0.25 layer height, the slicer preview shows the correct number of layer lines to get the 20 mm Z height.  And the display shows the correct number of lines being printed so the first 10 mm of Z height is being printed at slightly thicker layers than input or shown by the slicer preview.

### Expected results

I expect linear travel following Z step calibration of one mm travel for each one mm commanded.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7206011/cura.log)
[CE3PRO_20mm_calibration_cube.zip](https://github.com/Ultimaker/Cura/files/7206084/CE3PRO_20mm_calibration_cube.zip)
[CE3PRO_40mm_calibration_cube.zip](https://github.com/Ultimaker/Cura/files/7206085/CE3PRO_40mm_calibration_cube.zip)

GregValiant, 2021-09-21:
The stock Z stepper motor and lead screw assembly is 400 steps/mm.  If it isn't then it would seem to be either a driver or motor problem.  The lead screw moves an exact distance per revolution and the stepper moves an exact angle of rotation per step.  There are no pulleys that might have be out of tolerance and need tweaking.  It is the only thing on my printer (E3Pro) that I figure even Creality couldn't screw up.
So what did you end up with for steps/mm on the Z?

dn7050, 2021-09-23:
The Z steps are set at 400.1 after calibration that checked accuracy at 100mm and 200mm of Z travel.  This Ender 3 Pro has been converted to dual Z drive with two stepper motors (same as the original single motor) and each driven by its own TMC2209 driver on the SKR1.4 Turbo board.  The Z travel is accurate except that it adds in about 2.3mm by apparently slightly increasing layer height only over the first 10mm of Z travel.   Above 10mm, the layer height and Z travel is correct.  So either there is a gcode error in Cura in that first 10mm of Z travel or something in the SKR1.4 hardware or firmware (latest version of Marlin) is causing the extra layer height on the first 10mm.  It does not seem likely that it could be a mechanical issue with the Z stepper motors or drive screws.  It seems only be a coincidence that the error happens to be about the same as the Z offset of the nozzle relative to the BLTouch sensor homing position because when I set the offset to 0.0, the same thing happens with an extra 2.3mm added over the first 10mm of Z travel.  Of course that suggests the problem is probably in the SKR1.4 and its firmware rather than in the Cura generated gcode.

Ghostkeeper, 2021-09-28:
Your 20mm calibration cube results in g-code where the highest layer is reached at Z=19.88mm:

    G0 F300 X109.767 Y107.221 Z19.88

It's impossible to calculate this back from the initial layer height and normal layer height because Adaptive Layers is enabled. But suffice to say it's not nearly the +2.3mm you're seeing. It's a bit off in the other direction, even. It seems to be have 0.02mm of space for another layer at the maximum layer thickness of 0.2mm, but that may be an edge case that Adaptive Layers doesn't take into account.

You also tested manually moving the Z axis without Cura's involvement. So that pretty much rules out Cura as the cause of this issue, too.

So I'll close this as being a problem with your firmware rather than with Cura. I suggest you take it up with whoever wrote the firmware you put on that printer.

closed 2021-09-28 by Ghostkeeper