Ultimaker/Cura #10498: Non connected slices in new version

Jawood57, 2021-09-22:
### Application Version

4.11.0

### Platform

Windows 10

### Printer

Custom Creativity ELF Corexy

### Reproduction steps

Created a small cylindrical piece and sent to CURA. CURA prompted me to download latest version so I did. Send item to CURA, sliced and had a jagged white line on the rear after previewing.

### Actual results

Exported to CURA to try and it not print properly, it would go around and back skipping that area that was white on the slicer.

### Expected results

CURA should have continued on circle printing my item.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

nallath, 2021-09-23:
I don't understand what you mean, sorry. Could you attach the project file? 

The white dots are to indicate where the z-seam is.

Jawood57, 2021-09-23:
Here is the project file. This was made in Tinkercad. When trying to print the print head would not go around in a circle to build the cylinder, it would go around to the Z seam, stop and go back the otherway to the Z seam and reverse and go back the other way. It appears there was nor connectivity at the Z seam.

nallath, 2021-09-24:
Adding attachments via email doesn't work, sorry. You will need to go to the github web page and upload them there.

Jawood57, 2021-09-24:
https://www.tinkercad.com/things/bedlEspI928-jeffrey-project-001/edit?sharecode=fExLY3Oby34D-A37GBPbHCfKl43NDf0iZ3o-K3oIeLE
Not sure if this is what you want but lets try it. I am sharing a link to this project as it does not accept the file type  (See Below)

Jawood57, 2021-09-24:
I did attach with GitHub as a share I hope it works. This is new to me.

Ghostkeeper, 2021-09-29:
> https://www.tinkercad.com/things/bedlEspI928-jeffrey-project-001/edit?sharecode=fExLY3Oby34D-A37GBPbHCfKl43NDf0iZ3o-K3oIeLE Not sure if this is what you want but lets try it. I am sharing a link to this project as it does not accept the file type (See Below)

Sorry, but this is not a project file. With a "project file" we meant a Cura project. You can create it in Cura by going to File -> Save Project... This results in a file that contains your 3D model as well as all of the profiles and settings you're using. It allows us to precisely reconstruct your g-code and also see how Cura came to generate that g-code. Especially because you mentioned you customised your printer, it's important for us to know what's going on. I can't just load a 3D model into my own printer and expect to get the same result.

But yes, as mentioned before, the white dots indicate where on the outer wall the extrusion starts. This is usually visible as a blob in the print.
If nothing weird can be found in the g-code, I fear though that even a project file won't help us fix this for you. Seams are a print quality issue. I don't have your printer and don't get a significant seam with my own printer. All we can really control is Cura's output. If Cura's output is correct but the print is ugly, we can't really help to fix it. We need to know what we need to change in Cura's output that would fix it on your end.

As for general help with seams, here is an article from the Settings Guide that gives some hints on knobs you can tune in Cura to reduce the visibility of seams: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/troubleshooting/seam.md