Ultimaker/Cura #10499: I CANNT OPEN GCODE

CCTVSC, 2021-09-22:
### Application Version

4.10.0

### Platform

WINDOWNS 10

### Printer

COREXY

### Reproduction steps

1.OPEN GCODE FILES
2.ERROR

### Actual results

AttributeError: 'NoneType' object has no attribute 'close'
Thread 0x000047f8 (most recent call first):
  File "D:\Program Files\Ultimaker Cura 4.10.0\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 87 in _updateThread
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x000015e0 (most recent call first):
  File "threading.py", line 306 in wait
  File "zeroconf\__init__.py", line 2117 in wait
  File "zeroconf\__init__.py", line 1434 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00001340 (most recent call first):
  File "threading.py", line 306 in wait
  File "threading.py", line 558 in wait
  File "D:\Program Files\Ultimaker Cura 4.10.0\plugins\UM3NetworkPrinting\src\Network\ZeroConfClient.py", line 81 in _handleOnServiceChangedRequests
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00004010 (most recent call first):
  File "threading.py", line 306 in wait
  File "zeroconf\__init__.py", line 2117 in wait
  File "zeroconf\__init__.py", line 1262 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00003050 (most recent call first):
  File "zeroconf\__init__.py", line 1183 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x000034f4 (most recent call first):
  File "D:\Program Files\Ultimaker Cura 4.10.0\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 61 in _updateThread
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00001ce4 (most recent call first):
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-exe\build\inst\lib\python3.8\site-packages\UM\Backend\Backend.py", line 165 in _storeStderrToLogThread
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x0000423c (most recent call first):
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-exe\build\inst\lib\python3.8\site-packages\UM\Backend\Backend.py", line 153 in _storeOutputToLogThread
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap
[... 136 lines cut ...]
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-exe\build\inst\lib\python3.8\site-packages\cura\Arranging\Nest2DArrange.py", line 77 in findNodePlacement
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-exe\build\inst\lib\python3.8\site-packages\cura\Arranging\Nest2DArrange.py", line 126 in arrange
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 1931 in _readMeshFinished
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-exe\build\inst\lib\python3.8\site-packages\UM\Signal.py", line 332 in __performEmit
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-exe\build\inst\lib\python3.8\site-packages\UM\Event.py", line 218 in call
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-exe\build\inst\lib\python3.8\site-packages\UM\Qt\QtApplication.py", line 479 in event
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 1088 in event
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-exe\build\inst\lib\python3.8\site-packages\UM\Qt\QtApplication.py", line 393 in exec_
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 865 in run
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-exe\build\inst\bin\cura_app.py", line 233 in <module>
  File "Console.py", line 36 in run
  File "__startup__.py", line 74 in run

### Expected results

OPEN GCODE WITH CURA

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

nothing

nallath, 2021-09-23:
Does it work for 4.11?

Ghostkeeper, 2021-09-29:
I can't reproduce this error. Cura opens g-code files fine.

[image]

Perhaps you could share the specific g-code file that you tried to open?