Ultimaker/Cura #10500: crashing whilst slicing 3mf files, and arrange all items placing items beneath build plate

benjjyman, 2021-09-22:
### Application Version

4.11.0

### Platform

windows10

### Printer

geeetech A20m

### Reproduction steps

imported 3mf files after editing in fusion 360
multiplied both items by 1
clicked arrange all items
notice two parts were under build plate, so manually placed them
tried slicing twice then sent to this page

### Actual results

cura reports slicing with an unexpected error

### Expected results

usual just slice with same parameters provides with usable gcode

[cura.log](https://github.com/Ultimaker/Cura/files/7213431/cura.log)
[GA20M_y_axis_tensioner_slider.zip](https://github.com/Ultimaker/Cura/files/7213442/GA20M_y_axis_tensioner_slider.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[GA20M_y_axis_tensioner_slider.zip](https://github.com/Ultimaker/Cura/files/7213448/GA20M_y_axis_tensioner_slider.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/7213455/cura.log)

Vandresc, 2021-09-28:
Hi @benjjyman, yes this is indeed a bug and it is already on our radar. We will fix this in the next release. 
Until then we do have a workaround, change the Top/Bottom Pattern to anything than ZigZag or you can disable the Monotonic Top/Bottom Order. 
Thank you for your feedback!

Ghostkeeper, 2021-09-29:
Duplicate of https://github.com/Ultimaker/Cura/issues/10418.

closed 2021-09-29 by Ghostkeeper