Ultimaker/Cura #10501: Cura 4.11 (and for that matter, anything above 4.6.2) crashes when opening an STL file, Windows, Ender 3 Pro

rhubiere, 2021-09-22:
### Application Version

4.11

### Platform

 Windows 10 64bit

### Printer

Ender 3 Pro

### Reproduction steps

I've been having this problem with every version of Cura beyond 4.62.  Using Windows 10 64bit, Ender 3 Pro, whenever I open an STL file, Cura crashes.  No problems with 4.62, using the same exact settings and STL files.  Today I decided to add an 'ultimaker 2' printer (fictional, since I dont' have one) and sure enough, I'm able to open and slice the STL. While having the STL file open using the Ultimaker 2, I switched printers to the Ender 3 Pro & while it did not crash, I get an error that says "unable to slice".  I've read thru the threads regarding non-latin characters in the temp folders. While I don't have non-latin characters, I tried creating a temp2 folder at the root of C and changed both my temp & tmp variables...did not help. I also checked whether I have Chitubox (mentioned on another thread) but I do not have that installed.  Below is my log file.  Help...Please!  Thanks.

[cura.log](https://github.com/Ultimaker/Cura/files/7213901/cura.log)

### Actual results

Unable to open STL files, Cura crashes.  

### Expected results

Ability to open and slice STL files while having an Ender 3 Pro as my selected printer.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7213916/cura.log)

nallath, 2021-09-23:
It seems like your configuration got corrupted. We had an issue with 4.7.0 that could cause that. If you remove your 4.7.0 configuration (and upwards) it should re-create it from the 4.6.2 files that you still have. 

You can find the configuration folder in help-> show configuration folder. So first create a backup of the files there and then remove them. It should work again.

closed 2021-09-23 by nallath

rhubiere, 2021-09-23:
@nallath You are a scholar!  That did the trick.  Thanks so much.