Ultimaker/Cura #10502: don't slice model

cactus-ah, 2021-09-22:
### Application Version

4.11.0

### Platform

windows 10 64 bit

### Printer

ender 5 plus

### Reproduction steps

[image]
80% of the model slice goes forward and stops with this error

### Actual results

fail slicing prosses

### Expected results

in other ver don't exsist this bug

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

it just in 4.11.0 ver
[image]

nallath, 2021-09-23:
Please actually ad the log & project files. We can't help you if we don't have these files.

cactus-ah, 2021-09-23:
How can i send this file to you?

DragonFlyer007, 2021-09-23:
I have same error on latest version.

Ghostkeeper, 2021-09-29:
Previous versions of Cura didn't have this error because instead it would just let the progress bar freeze indefinitely. That message is new, indicating to the user that something went wrong.

To debug an error like this, we need reproduction steps that reproduce the error. The easiest way is to share a project file.

> How can i send this file to you?

You can create a project file by going to File -> Save Project... Put the resulting file in a .zip archive. Then go to the Github website (https://github.com/Ultimaker/Cura/issues/10502) and drag the zip archive onto the text box containing your response.

Attaching the file to an e-mail response doesn't work. Github ignores attachments to e-mail responses.