Ultimaker/Cura #10505: Will not slice model, error.

DragonFlyer007, 2021-09-23:
### Application Version

4.11.0

### Platform

Win 10

### Printer

Cr10S

### Reproduction steps

Tried to slice. First at 0.16 layer height, will not work.
Then tried 0.2 layers height. Did slice, but only one time. Tried many times after but not working.

### Actual results

Will not slice. just stalles after a long time, and then error message.

Log file will not upload?

project file i dont want to share really.

### Expected results

Expected to slice

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

nallath, 2021-09-24:
Without a project file, we can't help you out. We really need some steps to reproduce the issue. 

Is there another model that also has the issue that you do want to share?

DragonFlyer007, 2021-09-24:
> Without a project file, we can't help you out. We really need some steps to reproduce the issue.
> 
> Is there another model that also has the issue that you do want to share?

Im sorry but this is a model i sell, so dont want to share it it will be public? 
Have not had issues so far with other models but none in this version have been this complex either.

nallath, 2021-09-24:
> Im sorry but this is a model i sell, so dont want to share it it will be public?
> Have not had issues so far with other models but none in this version have been this complex either.

What I can offer is that you email the model to me privately. I promise that I will only use the model to figure out what this bug is / use it to reproduce it / fix the issue. You can send it to j.vankessel at ultimaker dot com

DragonFlyer007, 2021-09-24:
> What I can offer is that you email the model to me privately. I promise that I will only use the model to figure out what this bug is / use it to reproduce it / fix the issue. You can send it to [...]

Sent! :)

nallath, 2021-09-24:
The project file as you sent me works, but if I use the 0.16 dynamic profile, it doesn't slice. I originally thought it was the "use adaptive layers" but disabling that also makes it freeze. 

So I don't have a quick fix for you, but I will discuss it with the rest of the team.

DragonFlyer007, 2021-09-24:
Ok great.

Can you try it in prev version also?

I just deleted it, but might re install it then.

[https://ipmcdn.avast.com/images/icons/icon-envelope-tick-round-orange-animated-no-repeat-v1.gif]<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>  Virusfri. www.avast.com<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>

konskarm, 2021-10-04:
I tried running it in Debug mode, but I could not pinpoint the root cause of the issue. This needs more investigation and it's now in our backlog. Devs see ticket CURA-8622.

closed 2024-12-09 by github-actions[bot]