Ultimaker/Cura #10507: Can you guys please allow the notifications to disappear automatically?  Seriously...

rw3iss, 2021-09-24:
### Is your feature request related to a problem?

Sort of... constant changing or re-importing leaves lots of notifications (ie. if the object is auto-scaled) that just pile up and never go away/get in the way. They should disappear after a few seconds, or at least respond to an option to...  

### Describe the solution you'd like

Either have notifications automatically disappear, or do so from a configuration setting/toggle.

### Describe alternatives you've considered

Hmm, put the notifications in a scrolling view or clickable icon, or somewhere else that doesn't pile up on the screen... 

### Affected users and/or printers

Cura.

### Additional information & file uploads

[image]

nallath, 2021-09-24:
It's not a new feature, it's a bug! It should hide itself after 30 seconds, but for some reason it's not doing that for this specific message.

fieldOfView, 2021-09-24:
If this particular message is getting attention, I think it should refer to "model" instead of "object".

Ghostkeeper, 2021-10-01:
It seems that for some reason the QTimer that should auto-remove this message gets deleted by Python. Accessing its remaining time results in a segfault. The timer exists and is initially set to 30 seconds, but the timeout never triggers.

We don't have an idea on how to fix this, or why it only happens with this message. We've decided to not spend too much time on it now, in favour of other bugs.

rw3iss, 2021-10-01:
Haha, this is hilarious. I mean as a developer, one of those bugs and "screw it" mentality to just let it go, rather funny ;p 
... Hear you though, thanks. If I ever feel like compiling the code locally maybe I'll check it out ;)
I guess I'll leave this open as technically not fixed, but feel free to close it.

closed 2023-06-09 by github-actions[bot]