Ultimaker/Cura #10508: Bug

Sir-Knox, 2021-09-24:
### Application Version

Ultimaker Cura 4.11.0

### Platform

win 10

### Printer

Ender 3

### Reproduction steps

1. just install Ultimaker Cura 4.11.0
2, inportet STL- fil
3,i activate normal support
4. I started the print
5. i can see support is a 100% infill (WTF man)
6. and go's too zig zag when it gets close to the 3D print

### Actual results

support is a 100% infill

### Expected results

not "support is a 100% infill" but zigzag from the bottom all the way up to the 3D print

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Desktop.zip](https://github.com/Ultimaker/Cura/files/7224411/Desktop.zip)

nallath, 2021-09-24:
You set the support line distance to 0.4 So that is what you get; Support lines that are 0.4 mm apart (aka; the nozzle size).

Please check the tool-tips of settings carefully before changing them. If you want more information about what settings do, I can recommend the excellent settings guide plugin (https://marketplace.ultimaker.com/app/cura/plugins/Ghostkeeper/SettingsGuide2)

closed 2021-09-24 by nallath

Sir-Knox, 2021-09-24:
Thank you so much :)

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virusfri.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>