Ultimaker/Cura #10511: 4.11 Supports and Support Blockers not working

Raiden38, 2021-09-24:
### Application Version

4.11

### Platform

Windows 10

### Printer

Anet A8

### Reproduction steps

Added support, added support blockers

### Actual results

Some support are supporting nothing (there is no part over the support, it's just a tower of plastic with nothing on it. 
Support blocker show in gray, but they don't block anything in preview mode and in print.

### Expected results

Supports only when there is part to be supported
Supports removed when support blocker is active.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]
[image]
[image]
[image]
[Apple_TV_Mount.zip](https://github.com/Ultimaker/Cura/files/7227139/Apple_TV_Mount.zip)

Asterchades, 2021-09-26:
Try turning off the Use Towers option under Support. I've found it likes to generate little, superfluous towers when there's narrow objects which may (or may not) need support.

Support Blockers are to be used on where the supports are being generated _to_, not from. There's an area further up the model which the supports believe they're attaching to which is why they're still being generated.

Ghostkeeper, 2021-10-01:
Duplicate of https://github.com/Ultimaker/Cura/issues/7487.

closed 2021-10-01 by Ghostkeeper