Ultimaker/Cura #10512: 4.10 can slice this file; 4.11 cannot

snafu1282, 2021-09-25:
### Application Version

4.11

### Platform

Windows 10, Surface Pro 5

### Printer

Creality Ender 5 Pro

### Reproduction steps

1. load the attached model
2. try to slice the model

### Actual results

In 4.10 the model slices as expected. In 4.11, on one computer an error saying the slicing engine has crashed is presented. On another computer, the slice never completes. On that computer it is possible to Cancel the slice but it is not possible to complete it.

### Expected results

The model should slice as any other model does.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[Capture.zip](https://github.com/Ultimaker/Cura/files/7228924/Capture.zip
[cura.log](https://github.com/Ultimaker/Cura/files/7228926/cura.log)
[20mmx20mmx1x4x5.zip](https://github.com/Ultimaker/Cura/files/7228930/20mmx20mmx1x4x5.zip)
)

smartavionics, 2021-09-25:
I haven't looked at your project file but if you are using zigzag skin pattern and the new monotonic lines feature then this issue could be a duplicate of #10418.

snafu1282, 2021-09-25:
@smartavionics _Monotonic_ is not enabled and the top/bottom pattern is _lines_ (couldn't find anything called _skin pattern_, but I assume we're talking the same thing).

smartavionics, 2021-09-25:
Yeah, I meant top/bottom pattern. Oh, well, it was just a guess. I'll take a look at your project and see if I can find anything obvious...

smartavionics, 2021-09-25:
Ah, you haven't provided a project file, could you please do that (do File->Save and then zip or rename the .3mf file). Thanks.

snafu1282, 2021-09-25:
I believe I provided a project file. It's the attachment called 20mmx20mmx1x4x5.zip
I hope that's the proper way to include a project file.

smartavionics, 2021-09-25:
That file only contains the models and not the settings. Do File->Save not File->Export !

snafu1282, 2021-09-25:
I did _File>Save Project_. I do it for nearly all models. I will repeat it with a different filename & compare the 2. If the 2nd one is different I will post it.

snafu1282, 2021-09-25:
OK, here's a new file. I double-checked it was done using _Save Project_
[20mmx20mmx1_take2.zip](https://github.com/Ultimaker/Cura/files/7230224/20mmx20mmx1_take2.zip)__

smartavionics, 2021-09-25:
Ah right, it's already been reported, see #10394.

snafu1282, 2021-09-25:
Thanks. Good to know it's being addressed. I'm glad you know how to find out if an issue has been reported. I have no idea.

closed 2021-09-27 by nallath