Ultimaker/Cura #10513: Failed slicing

vallentx64, 2021-09-26:
### Application Version

4.11.0

### Platform

Windows 10

### Printer

Anet a8 plus

### Reproduction steps

Enabled monotonic top/bottom order, and slice it.

### Actual results

In the middle of slicing, the windows shows an app crash and then the slicing fail, but the app still running.

### Expected results

Some elevated tops drew monotonic.
[AA8P_grippv2 left.zip](https://github.com/Ultimaker/Cura/files/7230960/AA8P_grippv2.left.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

(this step normally works, but now the model a bit difficult)
(my English isn't perfect, sorry about that)

smartavionics, 2021-09-26:
Possibly same as #10418.

vallentx64, 2021-09-26:
Looks like it is, But not every time appears, maybe one flat tops isn't cause problem, but different directions, or heights.

Ghostkeeper, 2021-10-01:
Yes, this is #10418.

The issue occurs when the Zigzag pattern closes up the endpoints of the zag in a certain way and the code that groups up lines to be printed in monotonic order somehow ends up in an infinite loop. If the model is slightly shifted or rescaled or something it may stop occurring because the ends of the pattern close up differently.
We consider this to be a critical issue and it's been planned in for 4.12 but not fixed yet.

closed 2021-10-01 by Ghostkeeper