Ultimaker/Cura #10514: temperature tower reset after post processing

gamegine, 2021-09-26:
### Application Version

4.10.0

### Platform

Windows 10

### Printer

ender 3 pro

### Reproduction steps

hello, I have a bug to edit the temperature in post processing.
it performs the action then just after it resets the default value 

1 open stl and select profile: standard quality (0.2)
2 post processing > changeatz > set height and temperature
3 slice

on gcode he change m104 s215 and after he reset to m104 s220

### Actual results

```
G1 F1500 E310.9749
;[CAZD:
M104 S215.0 T0
;:CAZD]
M117 Extruder 1 Temp: 215
G1 F300 Z5
G0 F9000 X118.942 Y116.373 Z5
;TYPE:FILL
;[CAZD:
M104 S220.0 T0
;:CAZD]
M117 Extruder 1 Temp: 220
G1 F300 Z4.8
G1 F1500 E318.9749
```

### Expected results

```
G1 F1500 E310.9749
;[CAZD:
M104 S215.0 T0
;:CAZD]
M117 Extruder 1 Temp: 215
G1 F300 Z5
G0 F9000 X118.942 Y116.373 Z5
;TYPE:FILL
G1 F300 Z4.8
G1 F1500 E318.9749
```

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

project : https://cdn.discordapp.com/attachments/782742132078936074/891759432831238237/Heat_tower_190-240_Bridge.3mf

cura log : https://cdn.discordapp.com/attachments/782742132078936074/891764421611913276/cura.log

[image]

bremco, 2021-09-26:
I can confirm that this is a bug, and it's still present in the current release (4.11.0).

It seems that the script is confused by the retraction. Indeed, if retraction is disabled, it doesn't happen, so that's a viable workaround.

rburema, 2021-09-29:
The workaround is  even smaller, you don't have to disable retraction, but just z-hop, as that it the cause.

Unfortunately, we've decided to defer this for now.

closed 2021-12-06 by gamegine