Ultimaker/Cura #10515: After duplication of matrial profile it cannot be deleted

NilsRo, 2021-09-26:
### Application Version

4.11.0

### Platform

Windows 10

### Printer

Anycubic I3 Mega (but it is not relevant)

### Reproduction steps

1. Duplicated a material
2. Unlinked it
3. Try to delete it (button greyed out)
[image]

### Actual results

Button to delete the profile is not active even if it is not the active profile. The profile itself is unlinked.

### Expected results

If a material profile is not active it should be possible to delete it

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Tried with custom material profiles

bremco, 2021-09-26:
When I performed the reproduction steps (admittedly on Linux), the button was usable. Could you perhaps attach the logs? There may be some error/clue in there. (Or perhaps it's a graphical glitch: Is the one now shown as 'active' perhaps deleteable now?)

nallath, 2021-09-27:
A material that is active on any machine can't be deleted. Built in materials can't ever be deleted.

closed 2021-09-27 by nallath

bremco, 2021-09-27:
They're not trying to delete the built-in one, but the inactive clone
though.

nallath, 2021-09-27:
> […](#)

If you duplicate it, it also automatically activates it.