Ultimaker/Cura #10516: Material GUID is not changed after unlink custom material

NilsRo, 2021-09-26:
### Application Version

4.11

### Platform

Windows 10

### Printer

Every printer

### Reproduction steps

1. Duplicate custom material
2. Unlink new material
3. create GCode from both with material guid in start-gcode
4. GUID is the same in both GCodes even if the material is unlinked
[image]
[Gcode.zip](https://github.com/Ultimaker/Cura/files/7232137/Gcode.zip)

### Actual results

GUID is not changed after unlink custom material

### Expected results

A new GUID should be created after unlink a custom material

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Nothing

bremco, 2021-09-26:
Confirmed.

I think I can see where this is going wrong now. IIRC, the GUUID is generated using the (display) name, and a few other fields that supposedly create a unique key. This is done one creation, or in your case, the moment the unlink happens. As such, a change of name or other properties part of the key _after_ unlinking won't even fix this ... it has to be different from the original _before_ that (This is then also the workaround; rename before unlink.)

This, of course, is at the _very least_ confusing, and well, if it walks like a bug and quacks like a bug...

It's likely this is also why the other bug you reported (#10515) exists, and perhaps also why I couldn't reproduce it, since that's probably gone non-deterministic with identical GUUIDs (so will not reproduce 100% perhaps depending on system used).

fieldOfView, 2021-09-27:
> the GUUID is generated using the (display) name, and a few other fields that supposedly create a unique key

Er... no? It is generated with `uuid.uuid4()`; see
https://github.com/Ultimaker/Cura/blob/master/cura/Settings/ContainerManager.py#L373-L393

nallath, 2021-09-27:
> Confirmed.
> 
> I think I can see where this is going wrong now. IIRC, the GUUID is generated using the (display) name, and a few other fields that supposedly create a unique key. 

That's in the Print profile assistant. Not in "vanilla" Cura.

bremco, 2021-09-27:
Oops... Well regardless, a new GUID is only generated when the name is
changed, independant of the actual mechanism we use. At least that was what
I concluded from reproducing the priblem with and without name change.

nallath, 2021-09-27:
> Oops... Well regardless, a new GUID is only generated when the name is changed, independant of the actual mechanism we use. At least that was what I concluded from reproducing the priblem with and without name change.

If I duplicate a material and then press "unlink" and check the GUID that it stored on disk, it's different from the GUID from the material it was copied form, even if the name wasn't changed. 

It could be that it the GUID isn't stored correctly until a reboot of Cura is done though.

Ghostkeeper, 2021-10-01:
I could not reproduce it with the following steps:

1. Create a new Custom material.
2. Duplicate the Custom material.
3. Unlink the new material.
4. Make sure the first Custom material is selected. Slice a cube and save the g-code.
5. Make sure the second Custom material is selected. Slice a cube and save the g-code again.
6. Compare the diff of the two g-codes.

In the Griffin header the GUID is shown. A diff between the two g-code files showed a different GUID:

[image]

In the video of the OP it shows the two materials, but it still shows the "unlink" button for one of them. That's weird to me. Only one of the two materials has a unique GUID, meaning that the other one is still the same as some other material?

NilsRo, 2021-10-02:
@Ghostkeeper : I can also reproduce it with standard materials.

2.     Duplicate PLA generic material.
3.     Unlink the new material.
4.     Make sure the first material is selected. Slice a cube and save the g-code.
5.     Make sure the second material is selected. Slice a cube and save the g-code again.
6.     Compare the diff of the two g-codes.

[image]

I can confirm that the GUID is changed if the Display Name is changed only. The GUID duplication is also there after restarting Cura so the only workaround in 4.11 is to change the name. Perhaps it should be blocked to have materials with same name in general.

NilsRo, 2021-10-03:
HI everybody....I cannot delete the duplicated PLA anymore. :-) Show I create a second issue for this?

Ghostkeeper, 2021-10-04:
Okay, I was able to reproduce this error now, but the reproduction steps are a bit different.

1. Add Custom FFF Printer.
2. Change the start g-code of the printer to add a line at the end: `; {material_guid, 0}`
3. Load a cube (or any object really).
4. Duplicate Generic PLA.
5. Unlink the new PLA copy.
6. Make sure the new PLA copy is activated, slice and save the g-code.
7. Switch to the original Generic PLA (preferably with the Material Manager screen where you can confirm it's the original, since the original can't be edited).
8. Slice and save the g-code to a different file.
9. Compare the two g-code files.

The important bit is that it has to be done via the start g-code replacement keys. The Griffin header works correctly, but the replacement keys don't. Also, after restarting Cura it works correctly. Something seems to cache it or refuse to update. And it doesn't seem to be just the material GUID either, but every setting.

Ghostkeeper, 2021-10-04:
> HI everybody....I cannot delete the duplicated and renamed PLA anymore. :-) Show I create a second issue for this?

If you made sure the material you're trying to delete is not used in any extruder of any printer added to Cura, then yes.

Ghostkeeper, 2021-10-05:
We've discussed this issue and decided that we don't give this special priority over other issues. We're deferring it until it becomes a more serious issue. This decision was made because:

* It doesn't appear to be an issue with the printing temperature (which would be critical). It appears to be a missing update signal after the unlinking.
* It's only an issue with the start/end g-code replacement keys, not with the actual slicing settings.
* There are easy workarounds: Change any setting, or restart Cura after having unlinked the materials. These cause the GUID to update.

NilsRo, 2021-10-06:
> * There are easy workarounds: Change any setting, or restart Cura after having unlinked the materials. These cause the GUID to update.

In my case restarting Cura does not solve it. I have to change the name. To add this for documentation.

closed 2023-06-09 by github-actions[bot]