Ultimaker/Cura #10517: Fine Control of Cooling Fan Speed

blex-max, 2021-09-26:
### Is your feature request related to a problem?

I get the best mix of print quality and performance when running my cooling fan at about 50% of max, however dropping my fan to this speed gives me a little more ooze on travel moves. This isn't a problem, except on top surface skin layers where I tend to get unsightly blobs where I have a complex top surface that requires a few travels while doing the skin infill.

### Describe the solution you'd like

I'd like to be able to define fan speeds for layer types, e.g. top/bottom fan speed, or better yet top skin fan speed. I know I'm not the only person to have requested this but I wanted to add my voice to the many.

### Describe alternatives you've considered

I could tune retraction to try and minimise these blobs on top surfaces but I'm otherwise happy with my settings and I would rather have control of the fans.

### Affected users and/or printers

I run larger nozzles, which I have found are more sensitive to part cooling and as we all know have more ooze! I'm having this specific issue on a modded Ender 3 pro with a 0.6mm nozzle with some admittedly lower quality filament. Nevertheless, I think it'd be a valuable addition to Cura.

GregValiant, 2021-10-01:
This is similar to #10017 which is for fan control by layer as opposed to this request for fna control by "Type".  This could be added as an optional method.

Ghostkeeper, 2021-10-01:
More precisely even, I'm closing this as duplicate of https://github.com/Ultimaker/Cura/issues/5140.

closed 2021-10-01 by Ghostkeeper