Ultimaker/Cura #10518: Combing 'Not in Skin' still combs in skin

blex-max, 2021-09-26:
### Application Version

4.11.0

### Platform

Ubuntu 20.04 running on Thinkpad T480

### Printer

Creality Ender 3 Pro

### Reproduction steps

1) Sliced a calibration cube
2) Printed the cube
3) Noticed marks on top surface that looked like combing
4) checked to see what was happening in the slicer preview

You can see here that although 'Not within Skin' is selected for combing, it appears to be combing anyway.
[image]
[image]
[image]

### Actual results

Cura instructs the printer to perform combing moves on the top skin surface, around the Z of the calibration cube, leaving marks

### Expected results

Combing should have been disabled for the top skin surface, which usually results in the printhead moving via the open space of the Z in the cube, seen here
[image]
:

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Project File:
[combing_bug_cube.zip](https://github.com/Ultimaker/Cura/files/7232223/combing_bug_cube.zip)

smartavionics, 2021-09-26:
Sure it's combing, but not in the skin area, just along the walls...

[image]

nallath, 2021-09-27:
Combing everywhere
[image]

Combing "not in skin":
[image]

So as you can see in the "not in skin", it tries to stay in the walls as long as possible. You can see a good example of this in the bottom left corner: It first moves into the wall, then moves to the point where the distance to where it needs to go is the smallest and then it makes the crossing.

blex-max, 2021-09-27:
I appreciate that there's a difference if you run it with combing everywhere, so perhaps it's not quite a bug, but I was certainly surprised that not in skin operated in this way. Since the combing occurs at the end of skin infill extrusions, the resulting blobs do affect the skin even if it's mostly restricted to the walls.