Ultimaker/Cura #10520: Cannot create new material profile

gennartan, 2021-09-27:
### Application Version

4.11.0

### Platform

Manjaro 21.1.4

### Printer

Sidewinder X2

### Reproduction steps

1. Install Cura
2. Add new printer (sidewinder X1) with default parameter
3. Add new material profile --> (Ctrl-K   --> create)

### Actual results

`2021-09-27 12:20:02,802 - WARNING - [MainThread] UM.Preferences.getValue [105]: Tried to get the value of non-existing setting info/send_slice_info.
2021-09-27 12:20:02,835 - WARNING - [MainThread] UM.PluginRegistry._populateMetaData [778]: Could not find plugin global_stack
2021-09-27 12:20:02,836 - WARNING - [MainThread] UM.PluginRegistry._populateMetaData [778]: Could not find plugin extruder_stack
2021-09-27 12:20:02,844 - WARNING - [MainThread] cura.Settings.ContainerManager.getContainerMetaDataEntry [60]: Could not get metadata of container Unknown because it was not found.
2021-09-27 12:20:02,883 - WARNING - [MainThread] cura.Settings.ContainerManager.getContainerMetaDataEntry [60]: Could not get metadata of container Unknown because it was not found.
2021-09-27 12:20:02,896 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [421]: file:///usr/share/cura/resources/qml/Preferences/Materials/MaterialsDetailsPanel.qml:91: TypeError: Cannot read property 'container_node' of null
qml: TableViewSelection: index out of range
2021-09-27 12:20:04,775 - WARNING - [MainThread] cura.Machines.VariantNode.preferredMaterial [103]: Could not find preferred material generic_pla with diameter 2 for variant artillery_sidewinder_x1_0.4, falling back to empty_material.
2021-09-27 12:20:04,791 - INFO - [MainThread] cura.Machines.Models.MaterialManagementModel.duplicateMaterial [194]: Duplicating material empty_material to custom_material
`

### Expected results

See a new profile in the profile list, or an error mesage in cura explaining the issue.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I also tried to import a basic profile using the button "import". But the profile never showed up in the profile list.

Basic profile I tried:
`<?xml version="1.0" encoding="UTF-8"?>
<fdmmaterial xmlns="http://www.ultimaker.com/material">
    <metadata>
        <name>
            <brand>Generic</brand>
            <material>PLA and PLA</material>
            <color>Generic</color>
        </name>
        <GUID>506c9f0d-e3aa-4bd4-b2d2-23e2425b1aa9</GUID>
        <version>17</version>
        <color_code>#ffc924</color_code>
    </metadata>
    <properties>
        <density>1.24</density>
        <diameter>1.75</diameter>
    </properties>
    <settings>
        <setting key="print temperature">200</setting>
        <setting key="heated bed temperature">60</setting>
        <setting key="standby temperature">175</setting>
        <setting key="soften temperature">175</setting>
        <setting key="probe temperature">175</setting>
        <setting key="wipe temperature">175</setting>
        <setting key="part removal temperature">75</setting>
    </settings>
</fdmmaterial>
`

nallath, 2021-09-27:
I've tried to reproduce this with your reproduction steps, but I can't reproduce it.

If I add a sidewinder x1 and press crate material, a new material called "Custom Material" is added.

Could you perhaps share your configuration folder?

gennartan, 2021-09-27:
Here is my configuration folder.

[config.zip](https://github.com/Ultimaker/Cura/files/7236082/config.zip)

nallath, 2021-09-27:
Hmm, even with the configuration folder, I still can't reproduce the issue. 

I do see some things in the logs that give me the impression that you're not using the appimage. Is that correct? If so, it could be that something isn't set up correctly by whoever provided the build that you are using.

gennartan, 2021-09-27:
Yes, I was not using the app image. I installed cura using pacman.

I tried to use the app image and it works with it (but it still does not work without the app image).

closed 2021-09-27 by gennartan