Ultimaker/Cura #10521: Arrange All on Mac M1 does nothing

Vorchak, 2021-09-27:
### Application Version

4.11.0

### Platform

MacOS

### Printer

BIBO2

### Reproduction steps

It has been happening since 4.10. I have tried 4.10, 4.11 and 4.11 beta. All show the same symptoms. Load any STL or 3MF file and proceed to arrange all wither by right clicking on the object or Command-R and there is no action taken, i.e. objects remain untouched.

### Actual results

objects remain untouched

### Expected results

single object should move to the center of the build platform and on the ground.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]

nallath, 2021-09-28:
Could you also include the log files? It could be that there is something there that indicates what might be going wrong.

We officially don't support the M1 yet, but this is the first time that i've heard about the arrange all not working.

Vorchak, 2021-09-28:
I have cleaned up the logs and done the following:

   1. Started Cura (4.11 Beta but same issue for 4.10, 4.11)
   2. Loaded object
   3. Right click -> Arrange All
   4. Quit Cura

The full log is attached, but this is the section of the logs for the
Arrange All:

2021-09-28 08:19:04,333 - DEBUG - [MainThread]
> UM.View.GL.ShaderProgram.load [60]: Loading [/Applications/Ultimaker
> Cura.app/Contents/MacOS/resources/shaders/toolhandle.shader]...
> 2021-09-28 08:19:09,946 - ERROR - [JobQueueWorker [2]]
> UM.Logger.logException [112]: Exception: Unable to arrange the objects on
> the buildplate. The arrange algorithm has crashed.
> 2021-09-28 08:19:09,947 - ERROR - [JobQueueWorker [2]]
> UM.Logger.logException [116]: Traceback (most recent call last):
> 2021-09-28 08:19:09,949 - ERROR - [JobQueueWorker [2]]
> UM.Logger.logException [116]:   File
> "/Users/ultimaker/build/Cura/4.11/build/inst/lib/python3.8/site-packages/cura/Arranging/ArrangeObjectsJob.py",
> line 33, in run
> 2021-09-28 08:19:09,951 - ERROR - [JobQueueWorker [2]]
> UM.Logger.logException [116]:   File
> "/Users/ultimaker/build/Cura/4.11/build/inst/lib/python3.8/site-packages/cura/Arranging/Nest2DArrange.py",
> line 126, in arrange
> 2021-09-28 08:19:09,953 - ERROR - [JobQueueWorker [2]]
> UM.Logger.logException [116]:   File
> "/Users/ultimaker/build/Cura/4.11/build/inst/lib/python3.8/site-packages/cura/Arranging/Nest2DArrange.py",
> line 77, in findNodePlacement
> 2021-09-28 08:19:09,954 - ERROR - [JobQueueWorker [2]]
> UM.Logger.logException [116]:   File
> "/Users/ultimaker/build/Cura/4.11/build/inst/lib/python3.8/site-packages/UM/Math/Polygon.py",
> line 179, in intersectionConvexHulls
> 2021-09-28 08:19:09,955 - ERROR - [JobQueueWorker [2]]
> UM.Logger.logException [116]:   File
> "/Users/ultimaker/build/env/4.11/install/lib/python3.8/site-packages/shapely/geometry/base.py",
> line 676, in intersection
> 2021-09-28 08:19:09,957 - ERROR - [JobQueueWorker [2]]
> UM.Logger.logException [116]:   File
> "/Users/ultimaker/build/env/4.11/install/lib/python3.8/site-packages/shapely/topology.py",
> line 64, in __call__
> 2021-09-28 08:19:09,959 - ERROR - [JobQueueWorker [2]]
> UM.Logger.logException [116]:   File
> "/Users/ultimaker/build/env/4.11/install/lib/python3.8/site-packages/shapely/topology.py",
> line 18, in _validate
> 2021-09-28 08:19:09,960 - ERROR - [JobQueueWorker [2]]
> UM.Logger.logException [116]: ValueError: Null geometry supports no
> operations

I am also including the 3mf file just in case, although it does not matter
what file I load. Even a simple cube STL has the same result.

Thanks
   Hsiang

Ghostkeeper, 2021-10-04:
Actually we did get this reported a while ago for 4.10.

Duplicate of https://github.com/Ultimaker/Cura/issues/10133 .
I'm thinking it has something to do with the compiler we upgraded for 4.9. Not a lot of things changed with the arranger between 4.8 and 4.10.

closed 2021-10-04 by Ghostkeeper