Ultimaker/Cura #10523: Tree supports don’t cover as much area as regular supports

wjk, 2021-09-27:
### Is your feature request related to a problem?

I have a Prusa MK3S+ printer. The PrusaSlicer program shipped with it produces dense, up-and-down support structures. While these work, they take forever to build and are a pest to remove afterwards, especially when they touch the model on both the top and bottom.. Cura’s tree supports are much faster and far easier to remove, but with my specific printer, using them always results in a failing print. Consider the following image:

[image]

In this print, the tree supports does not cover the entire area of the bottom of the object. Every millimeter of the object with overhang past 55° must be supported or the print will fail, creating great clouds of curled-up spent filament. I can find no way to force the tree support to do this; it only supports part of the bottom of the object. Furthermore, there is always a small gap (no more than 2 or 3 layers) between the support and the object. This will also cause problems; the support and object must _touch_ or the print will fail.

### Describe the solution you'd like

Either change the tree supports to cover the entire bottom surface area of the object as with straight supports, or allow me to configure the straight supports to start from the build plate and angle out over the bottom of the model like tree supports do. I am perfectly fine with straight supports, but only if I can then easily remove them from the model afterwards. I have had to throw out several prints because I could not remove the supports from the nooks and crannies of the exterior.

### Describe alternatives you've considered

Consider the following screenshot, the same object sliced with straight up-down supports:

[image]

This print does indeed touch the bottom of the object (with Z Distance set to zero), and with the Enable Support Roof option set, every square millimeter of the object bottom is covered. This will print. However, these supports must also be placed on the top of the lower part of the model, which makes them harder to remove as the available vertical space to put the support decreases. (Some of my models have vertical gaps of only a few centimeters.) Tree supports can be configured to not touch the top of the model, but as shown above, I cannot configure them to have 100% coverage of the bottom of the object, even with Enable Support Roof set.

### Affected users and/or printers

I don’t know if it affects everyone with a Prusa MK3S+, because I’ve never asked anyone else who has one. Most people just use PrusaSlicer, but PrusaSlicer only prints dense up-and-down supports.

### Additional information & file uploads

Here is the project file I used to create the above screenshots:

[PI3MK3M_Overhang_Test.3mf.zip](https://github.com/Ultimaker/Cura/files/7236954/PI3MK3M_Overhang_Test.3mf.zip)

Ghostkeeper, 2021-10-04:
The X/Y+Z distance issue, where it keeps too much distance, is known for tree support. It's being tracked via https://github.com/Ultimaker/Cura/issues/8566. But it's not the main issue being reported here. Just keeping the discussion of that issue separate.

Tree support does go down to 55 degrees as specified by the settings, in your project file:
[image]

However the issue that I think you are trying to report is that it's only supporting most of it with a fairly thin branch:
[image]

Did I interpret that correctly?

The way it's implemented, Tree Support will mark a grid of "supported points" on the overhanging areas. These overhanging areas are often very thin slivers, because the previous layer supports part of the next layer and leaves only a few dozen micrometres unsupported if it's close to the overhang angle. The supported points are spaced out over a grid (with spacing Tree Support Branch Distance).
If an overhanging area ends up getting no branches to support it at all, one additional supported point is added in the centre of mass of that polygon (or the point inside the polygon closest to the centre of mass).

In your case, somehow it seems that for a long stretch of the overhang there are no grid points that fall onto the overhanging area. They apparently all generate in between the overhang areas, in the part of the layer that is supported by the previous layer (as determined by the support overhang angle setting). So you only have points in the centre of mass of each overhanging area, which is the centre line of the part in this case.

This issue has existed since Tree Support got implemented. It tends to happen with straight surfaces like this one, especially if the distance between layers matches the Tree Support Branch Distance. To fix this, we'd need to devise a way to spread the grid points of the tree's branches better over the thin overhang areas spanning multiple layers.

wjk, 2021-10-04:
> The X/Y+Z distance issue, where it keeps too much distance, is known for tree support. It's being tracked via #8566.

Understood. I’ll keep an eye on that issue.

> However the issue that I think you are trying to report is that it's only supporting most of it with a fairly thin branch. Did I interpret that correctly?

You’re right on the money. Given your explanation, it looks like I’ll need to find another model. Thanks so much for your help! 😄

Ghostkeeper, 2021-10-11:
I'd like to keep this issue open because it's a problem for many mechanical models. Sorry to overrule your closing! It is a real issue and we'd like to track it here.

Modifying the model is a workaround but not something we'd want to force on our users.

Hello1024, 2021-12-13:
> we'd need to devise a way to spread the grid points of the tree's branches better over the thin overhang areas spanning multiple layers.

Is there any requirement for the grid to be regular?  If not, why not just make a uniform pseudo random distribution of grid points in the overhang area, with the density determined by the config setting?

Obviously it isn't guaranteed to solve the problem (bad points could still be chosen by chance), but should eliminate pathologically bad models like this.

nallath, 2021-12-16:
This has been fixed for 4.13

wjk, 2021-12-18:
> This has been fixed for 4.13

Actually, no it wasn’t. Take a look at this screenshot (same file as OP):

[image]

The exact output has changed, but the problem remains exactly like it was before (not enough area, doesn’t touch the model no matter what I do). Please advise.

rburema, 2021-12-21:
@nallath Sorry, but this FR is a more broad-ranging topic than the bug that was fixed, see @Ghostkeeper's comments above. We'd like to keep it open.

rburema, 2021-12-21:
Ignore my previous remark, please.

closed 2021-12-21 by rburema