Ultimaker/Cura #10524: File Name Wrong on save to usb

1BlindSquirrel, 2021-09-27:
### Application Version

4.11

### Platform

windows 7 pro

### Printer

Ender 3 V2

### Reproduction steps

Imported a model, then sliced it.
Saved to USB.

BUT,
It was the wrong file-name & you can't know what file-name it will be because it don't ask for a file-name.
It overwrote the LAST file-name that was used!
That means BOTH files are corrupt!
The Only way to fix this mistake is to RE-NAME the mistake file-name.
Also Re-Slice the file-name that was corrupted!

This is a BIG MISTAKE!
Also happened on 4.9 on a friends cura yesterday.

### Actual results

BAD File-name!

### Expected results

Correct file-name!

Also ask user for file-name (This would eliminate corruption of existing files under the wrong name)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Nothing else to add

1BlindSquirrel, 2021-09-27:
FYI,
I contacted Cura Ultimaker about this problem.
They said because Cura (any-version) was USED for a product OTHER than Ultimaker It wasn't going to be addressed!

That's about the stupidest thing for a software brand to reply with I've EVER HEARD OF!
I'm hoping enough people that have the same problem STOP using Cura until this BIG PROBLEM can't be corrected.

Mine is:
Cura 4.11
Windows 7 Pro 64 bit

I know it's NOT a single incident, because it happen on another machine yesterday.
Cura 4.9
Window 10 64 bit

Hope This will generate enough interest from Cura Users...

nallath, 2021-09-27:
The save to removable drive is intended to be a quick convenience function. It uses the project name to quickly save a file to the removable drive. So at the least it's working as intended / designed.

> They said because Cura (any-version) was USED for a product OTHER than Ultimaker It wasn't going to be addressed!

I doubt it was worded like that. We do give priority to issues with Ultimakers, since all of the salaries of the developers are paid by Ultimaker. If there are issues with Cura that affect all people, we will do our best to fix it. 

>That's about the stupidest thing for a software brand to reply with I've EVER HEARD OF! 

Sorry, but you got this software for free. You didn't pay anything for this. So I don't feel that this response is warranted.

> I know it's NOT a single incident, because it happen on another machine yesterday.

Correct. It's by design; It's intended to be as quick as possible. This behavior has been like this since Cura 2.1. Not that that makes it "right" but I also couldn't find anyone reporting this before, so it gives me the impression that most people appreciate this functionality.

1BlindSquirrel, 2021-09-28:
1st, let me state:
I write 10+ programming languages and have written some that does hundreds of millions of calculations in seconds.
I've created my own websites (4 so far), and building another now.
  I think I know a wee bit about programming and naming conventions...
But you belittle me and try and make it like there is nothing wrong, or it's my fault.
_You must not know what you are talking about._

You are saying it's working as intended?
Did you NOT understand it was saving the .gcode as another project's name?
I deleted the email or I would have posted it for you (and others) to read (but I can go back to my server and check my saved files).
That's basically what was written in the email I got from Cura support!
Here it is again: "because Cura (any-version) was USED for a product OTHER than Ultimaker **It wasn't going to be addressed!**".

Yep it's free, and your saying **"it's OK to be making a mistake in the naming process?"**
And you wonder why I said the word STUPID?

Then you restate it's correct and making a mistake as designed?  (oh, I know hwat you said, but you completely skipped over the part that stated the software named .gcode wrong.

I can get those ,gcode files & .slt models and post them for all to see for themselves, even make a few .mpeg's showing the mistakes.
 **But I wasn't wanting this to become a pissing contest.**

I just thought (like most responsible and caring about others, kind of people), that others should watch out for naming convention mistakes.

If Cura don't care, then neither do I...

nallath, 2021-09-28:
> 1st, let me state: I write 10+ programming languages and have written some that does hundreds of millions of calculations in seconds. I've created my own websites (4 so far), and building another now. I think I know a wee bit about programming and naming conventions... 

Cool. I don't see how that changes anything though. 

> But you belittle me and try and make it like there is nothing wrong, or it's my fault. _You must not know what you are talking about._

I didn't say that, nor did I intend to say that. I merely pointed out that Ultimaker products have a preferential state for Cura, since that is what pays the bills. I was also trying to explain how (and why) we designed it as such. Seeing that you're a software developer yourself, you must be familiar with this. 

> You are saying it's working as intended? Did you NOT understand it was saving the .gcode as another project's name? 

No. Because you didn't state that. 

>That's basically what was written in the email I got from Cura support! Here it is again: "because Cura (any-version) was USED for a product OTHER than Ultimaker **It wasn't going to be addressed!**".

Please do. I don't think it was worded like that. If I am wrong about that, I would like to know who said that, so we can prevent it from being said like that again. 
 
> Yep it's free, and your saying **"it's OK to be making a mistake in the naming process?"** And you wonder why I said the word STUPID?

Yes, I wonder why you do that. But as I explained before; If you didn't buy an Ultimaker, I don't feel that you have any right to _demand_ anything from the Cura developers. You can kindly request something, but that isn't the tone I got from any of your messages. Statements like "Stupidest software brand" tend ruin a polite request for a fix, don't you agree? 

> Then you restate it's correct and making a mistake as designed? (oh, I know hwat you said, but you completely skipped over the part that stated the software named .gcode wrong.

No. I didn't. What i said is that the file in the removable drive is named according to the project name. You can set that in the bottom left corner.

> I can get those ,gcode files & .slt models and post them for all to see for themselves, even make a few .mpeg's showing the mistakes.

Please do. That would make it a lot easier to figure out what you mean, since I'm getting the impression that you mean something else than what I think you mean.

GregValiant, 2021-09-29:
@1BlindSquirrel, You have a Team Members take.  Maybe I can provide a Users perspective.

When a model is brought into Cura the name of the model file becomes the name of the project.  Bringing an additional model into the workspace does not change that initial project name.  If I were to delete that initial model from the workspace while leaving the second model, then the Project Name does not change even though the original model is now gone.  When I subsequently use the "Save to Removable Drive" function then that "obsolete" initial file name is used even though it isn't what I wanted.  It will over-write a file on the SD card (that has the same name) without any notification to the user.  In that way it is similar to using the QS command in AutoCad to perform a Quick Save.
Until @nallath mentioned it, I never knew that the Project name could be changed on the fly.  I live and I learn.
I've been scribbling code for a few days myself and I must say that the Project Name Change function isn't well marked in the window, but it is there.  For myself, I would file this under "know thy software" and is part of any users responsibility (although I would have made the Project Name Change box look different than the "Object List" that is right above it as (in my never humble opinion) it is certainly not intuitive or easy to notice).

[image]

I have found that saving a copy of the gcode on my hard drive just makes sense as I have had trouble with SD cards and Cura writes the file not knowing that it has written part of the file onto bad sectors.
Saving the file as a 3mf Cura Project allows for quick recovery of the model and the settings.  It's part of the "Department of Redundancy Department" rules I use when dealing with computers.  Back-ups are good.

On a point-by-point...
_"It was the wrong file-name & you can't know what file-name it will be because it don't ask for a file-name."_  I believe that the file name used is the Project Name as listed on the Title Bar of the workspace.  "Know thy software".
_"It overwrote the LAST file-name that was used!"_  Yep.  It's a no-frills Quick Save.
_"That means BOTH files are corrupt!"_  I don't get that.  The original file on the removeable drive has been over-written with the new file so I don't understand using with word "corrupt" in this context.  Yes, the first file is gone, but the second file should be fine, yes?
_"The Only way to fix this mistake is to RE-NAME the mistake file-name."_  Yes, that will work.  It will not get the first file back though.
_"Also Re-Slice the file-name that was corrupted!"_  I assume you mean the file that was over-written.  If there is a 3mf backup project then this is an easy thing to do.  "Know thy Software".

nallath, 2021-09-29:
>  "Know thy Software".

I think the "I didn't know I could do that" is a fair argument to make though! If the interface is unclear, it's also up to us to see if we can improve that.

Ghostkeeper, 2021-10-04:
The expected behaviour is that it uses the project name, as shown in the lower left corner, plus the appropriate extension for the toolpath format that your printer prefers (usually .gcode, but could also be .ufp or .x3g).

I think that 1BlindSquirrel expected the project name to be different in this case. The expected behaviour of the project name is:

* When loading the first model, the project name should take on the filename of that model (minus extensions). So when you load `cube.stl` the project name becomes `cube`.
* When loading another model while another model is still loaded, the project name should not be changed.
* When the last model is deleted and the build plate becomes empty, the project name should be cleared. Loading a new model will cause it to then take on the name of that model.

So what could have happened is something like:

1. 1BlindSquirrel loaded in one model e.g. `cube.stl`. The project name is then `cube`.
2. They loaded in a second model e.g. `pyramid.stl`. Since it's the second model, the project name remains `cube`.
3. They deleted the `cube.stl` model. Since the build plate doesn't become empty, the project name remains `cube`.
4. They saved the g-code to removable drive, which then becomes `cube.gcode` even though the only remaining model is `pyramid.stl`.

Did I get that right, @1BlindSquirrel, or did something different happen? What would you want to behave differently?

1BlindSquirrel, 2021-10-04:
Ghostkeeper,
  You got it wrong.
a) I **didn't** create a PROJECT NAME.
b) I opened cura 4.11, then I opened a file named cube.stl and sliced it & saved it to disk (the micro SD card).
c) I deleted that model & the saved popup message.
d) I opened another model named another.stl, sliced it & saved it to disk (the micro SD card).
e) I delete that model & the saved popup message.

Now I'm ready to open the 3rd .stl to slice & save.

  Just being curious,
I was going to open the 2nd .gcode in cura 4.11 to make sure it was correct, it wasn't there at all.
The 2nd model's name was cube.gcode (not another.gcode like I would have thought).
Wait!
That was the wrong filename.
I opened it in cura 4.11, indeed it was the wrong model for the filename.

  I opened the 1st model (by time/date attributes), it was named whatever the model's name that was sliced previous to the 1st cube.stl...

I re-opened, sliced & saved them both again (see b,d above), but the same thing happened again.
The only difference was the filenames shifted to the filenames that were sliced/saved 1 filename previous.
 They were wrong filenames again!

Since, I've gotten correct slice/save .gcodes, but they still save to wrong filenames every now and them (at random, no pattern I can discern).

**Update:**
  I just sliced/saved 39 .stl's.
The models are part of a BIG object that will be glued together.
All parts are to be printed with specific colors (6 different colors of filament, folder is color name).

Now I have to go back and check each & every .gcode to make sure they are the correct part with the correct filename in the correct color folder.

  That's a lot of work because the naming convention isn't TRUSTWORTHY!
I hope this is plain enough for everyone to understand.

fieldOfView, 2021-10-04:
> If the interface is unclear, it's also up to us to see if we can improve that.

Originally, the project name was in the lower-*right* corner, next to the save button area. That way there was more of a link between the name and saving the file.

I think I am going to move it in with the save button area in the Sidebar GUI plugin, or at least move it back to the right instead of the far left.

nallath, 2021-10-04:
> Ghostkeeper, You got it wrong.
 a) I **didn't** create a PROJECT NAME. 
 
 Aaah. See, we found the issue. As quoted from the steps from ghostkeeper:
 
 > When loading the first model, the project name should take on the filename of that model (minus extensions). So when you load cube.stl the project name becomes cube.

GregValiant, 2021-10-04:
_"I think I am going to move it in with the save button area in the Sidebar GUI plugin, or at least move it back to the right instead of the far left."_
Where it is on the left is actually an uncluttered area.  It's just that the single line with a cryptic icon made it appear (to me) that it was part of the Object List.  Admittedly I didn't look any closer (as I'm old and lazy) but a label would be nice and maybe a larger font to differentiate it from the Object List?

Ghostkeeper, 2021-10-05:
> b) I opened cura 4.11, then I opened a file named cube.stl and sliced it & saved it to disk (the micro SD card).
c) I deleted that model & the saved popup message.
d) I opened another model named another.stl, sliced it & saved it to disk (the micro SD card).
e) I delete that model & the saved popup message.

> I was going to open the 2nd .gcode in cura 4.11 to make sure it was correct, it wasn't there at all.
The 2nd model's name was cube.gcode (not another.gcode like I would have thought).

That would indeed be a problem! Since you cleared the build plate in step c, it's expected that the print job name would be set to "another" when loading `another.stl`. After all, it's then the only object loaded on the build plate.

Repeating these steps, I'm getting that the first file is saved as `CFFFP_cube.gcode` and the second file as `CFFFP_another.gcode`. So I'm unable to reproduce this bug then. For me it's working as expected. Strange. There must be something else going on that's different on your computer than on mine.

I think the job name was better placed on the right for the flow. As I remember, the original flow was designed with the progression in mind that the user would start from the top left, work towards the right and then down towards the print button. But I agree with the current design that it would look weird to have the name next to the print button since that button is now floating. It would work as long as the text is not taller than that action panel, but since the object list can fold out it would be overlapping with the settings list. And moving _only_ the job name to the right (leaving the rest on the left) would look a bit weird too I think.

1BlindSquirrel, 2021-10-05:
Ghostkeeper,

  I remember looking at the filename on the left side (by the way in my opinion, it's great where it is), and thinking "That's the correct filename".

Also,
  Remember I said this happened on a totally different machine, with cura 4.9
I used the exact same steps I outlined above (a,b,c,d,e).
I asked the owner to save that file for me, just haven't had the time to run by and get it yet.

Yes it's very strange.
I just originally wanted to let others know of this potential naming mistake, but it has turned into something like a bashing instead, where I'm having to defend what I stated.
It happened.
I won't make that mistake again...

Ghostkeeper, 2021-10-06:
Using phrases like "that's the stupidest reply I've ever heard of!" right off the bat and capitalised words don't exactly invite an intellectual discussion. Glad we can put that behind us though and focus on the issue you're having

As far as I'm concerned there are two issues here. One is that contributors realise that it might not be clear where the name of the file is coming from, aside from 1BlindSquirrel who said it was okay. The other is the original issue of 1BlindSquirrel.
However the original issue cannot be reproduced by us so far, not by executing the reproduction steps b) through e) outlined [here](https://github.com/Ultimaker/Cura/issues/10524#issuecomment-933531710).

I wouldn't think this is computer-specific as it's all in our own code, hardly to do with Qt. But it's possible. We could wait and see if other people encounter the same issue. If they do, it's important that we narrow down the reproduction steps correctly.

One other thing for @1BlindSquirrel to check: Do you have plug-ins installed that affect the print job name? These include:

* Gcode Filename Format
* Gcode Filename Format Plus
* Printjob Naming

Those could be interfering with this behaviour, perhaps.

1BlindSquirrel, 2021-10-11:
I have no other plug-ins installed.
Did you miss another machine with a different operating system had the same filename error too?

I haven't had the time to get the gcode off the other machine.
I also didn't go back over the 39 files I sliced and saved either, but it will be interesting to see what happened.

dmitche3, 2021-10-19:
What you are describing is the "open file" process.  Yes, if you have an item loaded into Cura and you add another file it is suppose Wow.  That is just lame and stupid.  If you are replacing the contents of the build plate than why keep a meaningless name that has no correlation to the object that is on the build plate?

nallath, 2021-10-20:
If you replace the objects on the build plate (eg; _clear_ the buildplate and then load another file) it will use the name of the last loaded object (and not the first one).