Ultimaker/Cura #10525: Part is missing sections after slice

ZTASecurity, 2021-09-27:
### Application Version

4.11.0

### Platform

Windows Pro 11

### Printer

CAMBRIAN TPR - Added using the Atomstack files

### Reproduction steps

The part shows up correctly in the preview but when the model is sliced it is deleting part of the part. Please see the attached screen captures of the model and the model preview
[image]
[image]
. 

### Actual results

The part prints as shown in the preview

### Expected results

The part should have printed as shown in the model view

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CFFFP CFFFP Carabela Hand Grip Rev 1C AtomStack TPR lw 0.8mm lh 0.2mm if 100% ext1 200C bed 60C AtomStack TPR lw 0.8mm lh 0.1mm if 100% ext1 200.0C bed 60.0C.3mf.zip](https://github.com/Ultimaker/Cura/files/7238105/CFFFP.CFFFP.Carabela.Hand.Grip.Rev.1C.AtomStack.TPR.lw.0.8mm.lh.0.2mm.if.100.ext1.200C.bed.60C.AtomStack.TPR.lw.0.8mm.lh.0.1mm.if.100.ext1.200.0C.bed.60.0C.3mf.zip)

GregValiant, 2021-09-27:
That .8 nozzle is too big for those fine details.  Drop to a .4 and the circles will show up.

nallath, 2021-09-29:
I'm closing this, since the answer of Greg is correct :)

closed 2021-09-29 by nallath

ZTASecurity, 2021-09-29:
This is still an open issue. My seconds question has not been answered.

Thanks
Pauk

www.ztasecurity.com

Closed #10525<https://github.com/Ultimaker/Cura/issues/10525>.

nallath, 2021-09-29:
> This is still an open issue. My seconds question has not been answered. Thanks Pauk [www.ztasecurity.com]

Second question? I only see you asking why those parts weren't printed. That is caused due to the details being too small for the nozzle that you're using.

ZTASecurity, 2021-09-29:
I adjusted the feature size to be a multiple of 0.8mm and that improved the output of the slicer. I can change to a 0.6mm or 0.4mm nozzle however I have to check with Atomstack if these are available for this extruder. The material is 2.85mm TPR.

Attached are pictures showing the current preview with the model adjusted. Can you explain what the white marks represent?

Thanks
Paul

www.ztasecurity.com

nallath, 2021-09-29:
The white marks are places where the extrusion stopped (or started). These are places where you tend to get small scar / artifacts. As such, we display them in the interface (also because there are quite some settings that influence the position of these marks)

ZTASecurity, 2021-09-29:
Thanks Jaime,

Please close this issue.

Paul