Ultimaker/Cura #10527: Draft Shield Speed Control

efnikolas, 2021-09-27:
### Is your feature request related to a problem?

From what I understand, the speed of the draft shield follows the skirt/brim speed, which seems a bit wrong in my opinion since it can definitely be faster when it's not on the first layers.

### Describe the solution you'd like

I would like to have separate speed control for the draft shield. 

### Describe alternatives you've considered

Controlling the speed with the brim doesn't work because it slows the print too much without any apparent reason since it doesn't need extra adhesion etc.

### Affected users and/or printers

Artillert Sidewinder X1 
Creality Ender 3 
Every 3D printer, if the user decides to use the draft shield

Ghostkeeper, 2021-10-04:
Or it could be printed with the Support Infill Speed, being something that needs to stand stable but nobody really cares to tune the speed/acceleration/jerk of the draft shield.

aleph00, 2026-09-24:
Hi guys !
This issue is quite old, but I ran into the same problem lately and didn't find any improvement on it since 2021. So, with the help of chatgpt as I'm not a developper, we wrote a post-processing script to change the speed of the draft (and only that). You can find it on my github : [Cura-DraftShieldSpeed](https://github.com/aleph00/Cura-DraftShieldSpeed)