Ultimaker/Cura #10538: Cura Crashes with UM3 extruders unenabled or when not as printer

sonicdudemg, 2021-09-28:
### Application Version

4.11/4.10

### Platform

Windows 10

### Printer

Ultimaker 3

### Reproduction steps

1. Unenable printer core 1
2. switch printers to another one in the list.
3. Program crashes

### Actual results

the program immediately crashes. Gives error report and closes.

### Expected results

the expected results should be that the printer core 1 is not enabled and the printer can switch to the other printer with no problem.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7247856/cura.log)

sonicdudemg, 2021-09-28:
May possibly be for incorrect material too.

rburema, 2021-09-29:
I can't reproduce that in vanilla Cura... could you try to disable the SmartSlice plugin, and do the reproduction steps?
I see it throwing some errors in the logs.

sonicdudemg, 2021-09-29:
I disabled it and it gives me an error.[Image.jpeg][Image.jpeg]

rburema, 2021-09-30:
I'm sorry, it seems that the image you wanted to share didn't attach properly. I think it might be that attaching an image only works when you're on the github site itself (just drag the image to the post), not via mail?

Ghostkeeper, 2021-10-05:
Please report this issue with the SmartSlice plug-in: https://github.com/tetonsim/is-cura-ui/issues

I think they're mixing up some types. It appeared to put a formula somewhere in a setting property which should have a simple value.

closed 2021-10-05 by Ghostkeeper