Ultimaker/Cura #10543: support and and skirt is are not respecting machine_disallowed_areas

moham96, 2021-09-29:
### Application Version

4.11.0

### Platform

windows 10

### Printer

Ender-3

### Reproduction steps

1- to show the bug use a printer with a defined `machine_disallowed_areas` it's the Ender-3 in my case
2-add a model and position it so it is on the edges just before the disallowed area
3- enable skirt and/or tree support
4- after slicing you will see that the support and the skirt is allowed to be in the disallowed area

### Actual results

The support material and the skirt are allowed to be in the disallowed area

### Expected results

The support and the bed adhesion material should not be allowed in the disallowed areas

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CE3_rfid_case.zip](https://github.com/Ultimaker/Cura/files/7253392/CE3_rfid_case.zip)

[image]

Ghostkeeper, 2021-10-06:
The problem here is only with tree support and conical support, not with the skirt. And I realise indeed that this is a potential problem with the custom disallowed areas.

So to clarify: Cura has two types of shadowy areas, if you will. We tend to mix up the terminology ourselves too, but to disambiguate we should call them "custom disallowed areas" and "disallowed border". The Ender 3 has both: It has custom disallowed areas to denote where the clips holding down the build plate could be placed, specific to this printer due to its hardware. And every printer has a disallowed border which goes all around the build volume and around every custom disallowed area, which is meant to keep space for things like the skirt.

In other words, related to the screenshot above: It's allowed to go into the lighter-grey shadow area because that's the region that Cura allocated itself for the skirt. Without tree support or conical support, it shouldn't ever go into the darker-grey shadow area.

For the skirt this works fine. The skirt itself will intersect with the disallowed border. That's what it's meant for after all. This is okay. It can't intersect with the custom disallowed areas because the disallowed border is also drawn around custom disallowed areas, so that the skirt will also not hit those clips.

The problem is with tree support and conical support. For those it's unknown where the support will end up until slicing is complete. And it could expand wider than the model itself. So it could decide to place tree support outside of the build volume. For that I made an [invisible collision area](https://github.com/Ultimaker/CuraEngine/blob/f302fa86b3b179b1a1e191cf31bcee93e8a1f512/src/TreeSupport.cpp#L730-L737) around the whole printer, to prevent the tree support from expanding beyond the size of the printer. This makes use of the total size of the printer.

However: **this doesn't look at the custom disallowed areas**. It only looks at the total build volume, keeping enough space around it at the bottom for the skirt/brim/raft.
This is the real problem I believe you're encountering here.

Based on this, I'd expect that you can't reproduce this problem in the X borders of the printer. There is a disallowed border there, but no custom disallowed areas (because apparently the clips don't get placed on that side).

fvrmr, 2021-10-15:
We have discussed this issue with the team and we decided to defer this issue. Since it is a bit of an edge case and only with tree and conical support. So it is not a priority for us to fix this. I hope you understand.

tastyratz, 2022-02-04:
I experienced this today as well and wanted to comment on it to follow and contribute. I don't have an ender, this was on a Wanhao Duplicator 6. I see the skirt drawn outside of the acceptable boundings when I print but I also get warnings in Octoprint

Object in WD6_Side_left_bottom_lefttop_rightbottom.gcode exceeds the print volume of the currently selected printer profile, be careful when printing this.
More
Object exceeds print volume in depth.
Object's bounding box: (2.00, -5.80, 0.00) × (198.26, 196.20, 8.40)
Print volume: (0.00, 0.00, 0.00) × (200.00, 200.00, 180.00)

closed 2023-06-09 by github-actions[bot]

RevHarryPowell, 2023-12-25:
This is still a problem in version 5.5.0.  I defined this disallowed area for a bed clip.  Cura prevents me from placing the model there, but it will still generate a skirt in the area.

[image]