Ultimaker/Cura #10544: Cura 4.11 printing inefficiently and too fast, leaving a bad finish in the model.

Ivomola, 2021-09-29:
### Application Version

4.11.0

### Platform

Windows 10

### Printer

Creality Ender 3

### Reproduction steps

1. I updated to Cura 4.11, enabled monotonic top and bottom pattern and increased first layer line width to 120%.
2. I imported the attached model and sliced it.
3. Done.

### Actual results

The printer prints inefficiently and too fast (see attached files).

### Expected results

The printer should follow one path smoothly and print in an efficient manner instead of switching sides pointlessly (did not happen in previous versions of Cura)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

The model, project file, video to show how bad the Gcode generated is and videos and pictures of the print can be found here: https://mega.nz/file/RlgFBIgJ#UlnqGSW4y70Q0hDylsKjBsbf9HXJCq79xA-QI4IT5AI.

nallath, 2021-09-30:
The "switching pointlessly" is actually needed for monotonic fill. 

For this model I would also look at the Arachne version of Cura; it should do a much better job.

smartavionics, 2021-10-02:
> The "switching pointlessly" is actually needed for monotonic fill.

Indeed, monotonic skins can require more travel moves than used when printing the same skin non-monotonically. However, in this example UM's Cura is doing a lot more travel than it really needs to. Using the UM monotonic implementation, it's showing 23 mins to print.  Using an alternative monotonic implementation as can be found in https://github.com/smartavionics/Cura/releases it takes 15 mins to print.

Ghostkeeper, 2021-10-06:
One shortcoming of the current Monotonic implementation is that it orders segments only in the monotonic direction. Within the constraints of Monotonic there is still freedom to order segments differently. Cura chooses one valid monotonic ordering, but a different monotonic ordering could have shorter travels.

An example where this goes wrong is this test I made during the implementation:
[image]
Since the teeth of that model are all at equal position along the monotonic direction, it chooses the teeth arbitrarily based on rounding errors. This means that the nozzle goes back and forth a bit.

However this was taken out of scope of the original implementation. We have a ticket about it in our backlog already: CURA-8474.